ma1co/OpenMemories-Tweak #1: Contents of Backup.bin

damircolak, 2016-04-25:
Could you please make available the backup.bin of A7RII or any other Sony camera?

That way we can help you analyze it and see what else can be done.

ma1co, 2016-04-25:
Use [fwtool](https://github.com/ma1co/fwtool.py) to unpack your camera's firmware update, the backups are in the *0110_backup* folder (different files for different regions).

This reminds me that I should probably document the format somewhere, haven't done this yet. Look at */usr/kmod/backup.ko* for the driver, it's not too hard to figure out what it does.

nkhdk, 2016-04-25:
Would you be interested in collecting the _backup.ko_ files from various models/regions, or does that not really matter?

damircolak, 2016-04-25:
Having a repository of backup.bin files, one for each camera model would be enough, I imagine.

Boldbrew, 2016-04-26:
It could be better to translate them into text files and compare side by side. No property names but only id's. But even this may be a good first step. Is not it?

incrediblemac, 2016-04-26:
I am a little confused here. Sorry for my ignorance.. Where can i find this _backup.bin_ file ? and How to read/decode these .bin files?

Unpacked the latest A6000 firmware ( update 3.10) . I can't find a file named _backup.bin_ in _0110_backup_ folder. Instead i have a _SYSMUSASHI-DSLR_ folder and _MainDest.txt_ . There is no backup.bin under that sub folder as well. However there are some other .bin files with name starting with _CX7750_ for eg _CX77500_ALLLANG.bin_

ma1co, 2016-04-26:
These are all the `Backup.bin` files for different regions (minus the settings that are already stored on the camera, like the serial number). I haven't documented the file format yet. Have a look at `backup.ko` if you want to reverse engineer it yourself.

lh8511, 2016-04-26:
no backup.ko on Nex6 firmware... Must be another driver!

incrediblemac, 2016-04-26:
Ok. Where can i find `backup.ko` ? `/usr/kmod/backup.ko` ? I don't have such a folder in firmware.tar_unpacked  or updater.img_unpacked folders.

damircolak, 2016-04-26:
Could you please attach the backup.ko here so that we can analyze it without having to extract it from a camera?

ma1co, 2016-04-26:
Yap, for 2nd gen devices, it's *unidrv2.ko* in *initrd.img*.

And no, we won't distribute any code by Sony here.

podmanvn, 2016-04-26:
So...no backup.bin for A7M2 Japan-only ? Maybe it already changed into backup.ko ?

ma1co, 2016-04-26:
The Japanese-only backup file is included in every firmware. The filename ends with *_J1.bin*

Boldbrew, 2016-04-27:
// Have a look at backup.ko
No any names and/or hints in it regarding the meaning of the settings. It's only message receiver. At least i could not find anything it it. Main question for me is how you've able to find 0x3E0166 message type. But i understand that such information probably will not be published. In other hand it's probably be a bit easy to do some boring job in data collection and summarizing by somebody else but you. 
Anyway your job is excellent! Thank you very much.

Boldbrew, 2016-04-27:
I hope this could help to somebody with other command research http://pastebin.com/cQH9x4z9

lainy, 2016-04-27:
@Boldbrew thanks, that is helpful. I assume you extracted it from adjstctl.elf in nflasha15? At least, that's where I found it in rx100m4 firwmare v1.30. Based on the function that reads that table and dumps it with fprintf, I've added headings: http://pastebin.com/NY6uP15C

Boldbrew, 2016-04-27:
@lainy Most probably. But first fields used as bytes is not it? I mean 0x1, 0x1, 0x10, 0x1. And not all the meaning is clear for me. Anyway they pretty useless without clear payload data forming understanding.

lainy, 2016-04-27:
@Boldbrew They are used as 16bit values, like:
```c
fprintf(stdout, "adjust block: 0x%04x, code: 0x%04x OSAL ID: 0x%08lx cmdname: %s\n", block, code, osal_id, cmdname
```
Here is the relevant snippet:
```asm
loc_8A5A                          ; r4 is pointer to *next* struct
  LDR.W           R1, [R4,#-4]    ; so -4 is last 4bytes of current struct (set r1 = osal id)
  LDRH.W          R2, [R4,#-0x48] ; sizeof(struct) is 0x48, so r2 = block (first 16bits of struct)
  LDRH.W          R3, [R4,#-0x46] ; r3 = next halfword (code)
  STR             R1, [SP,#0xC+osal_id] ; store osal_id on stack
  MOVS            R1, #0x48
  MLA.W           R1, R1, index, R6 ; r1 = 0x48 * index, r6 not used
  ADDS            index, #1
  ADDS            R1, #4  ; offset of cmdname
  STR             R1, [SP,#0xC+cmdname_ptr] ; store cmdname ptr on stack
  LDR             R1, =stdout
  LDR             R0, [R1] ; stream
  LDR             R1, =aAdjustBlock0x0 ; "adjust block: 0x%04x, code: 0x%04x OSAL ID: 0x%08lx cmdname: %s\n"
  BLX             fprintf ; r0 is stdout
                          ; r1 is format string
                          ; r2 is block (16bit)
                          ; r3 is code (16bit)
                          ; stack contains osal_id (32bit) and cmdname (char*)
```

But you're right, not much use without a better understanding of it all. I've done some poking of my own on the Sony firmware to learn about the lens protocol, but always got distracted with other things...

Anyway, I suspect the data we want is in av-cam.bin (nflasha3), as it seems to handle the main camera functionality, but I could be way off :sweat_smile:

lainy, 2016-04-27:
Well, the movie record time stuff seems to come from libmpr.so I think. At a glance, I see the hour, minute, and second values being referenced in a function which calls Backup_read() with them.

Boldbrew, 2016-04-27:
Looks so. A number of BKID_* constants could be found around. But they're not much promising from my point of view.

MarcBlast, 2016-05-04:
Hi guys, thank you so much for contributing to this project! I can't wait to pitch in my help and expand the capabilities! Can you share the method of decoding the binaries? I'm using Hexinator right now looking at adstctl.elf and I recognize some phrases. But how can I convert them into the snippet that you shared in pastebin? Thanks!

lainy, 2016-05-04:
@MarcBlast I'm using [IDA Pro](https://hex-rays.com/products/ida/) to analyze the files, but a cheaper alternative is [Hopper](http://www.hopperapp.com/). There is also [r2](http://radare.org/r/), but I've never used it.

gtoni48, 2016-05-10:
Hi guys, sorry my less experience. I have a problem to search the Backup.bin file from my a6000 update. I have a 3.10 update, and i do uncompress the update with "fwtool" but just come back two folders (nexhack and resource) and another update installer. Where is the file Backup.bin? Whats wrong with my worksop for uncompress the update file? 

Sorry for my english, im from Spain. Thanks for help!

ma1co, 2016-05-10:
@gtoni48 Look at the 2nd comment in this thread.

lh8511, 2016-05-10:
Hola @gtoni48 ,
- Are you using this fwtool (http://github.com/ma1co/fwtool.py)?

Once unpacked, you will have 2 folder for A6000:
         updater.img_unpacked
         firmware.tar_unpacked   -- Firmware files are here

"Backup.bin" files : firmware.tar_unpacked\0110_backup\SYSMUSASHI-DSLR\DM_UUD3

gtoni48, 2016-05-10:
Thanks, now i have fwtool.py from your link. But, how can i execute the software? Im really sorry and thanks again!

lh8511, 2016-05-10:
You must install python and then run from command line

python fwtool.py unpack -f FileUpdateNamexxx.exe

Bostwickenator, 2016-05-10:
Before that command you'll need to run this one "pip install pycrypto"

ma1co, 2016-06-11:
I've added an option to [fwtool.py](https://github.com/ma1co/fwtool.py) to print all properties contained in a Backup.bin. There are now binaries for Windows and OS X, too.

closed 2016-06-11 by ma1co

lh8511, 2016-06-12:
@ma1co Thanks for print_backup command :-)

AlexDaniel, 2019-11-19:
@ma1co thank you for your efforts!

So I opened my Backup.bin file with GIMP. Does anybody know who is that guy?

[image]

AlexDaniel, 2019-11-19:
I did the same with another camera of the same model (AX100E) and found no face. Could it be some kind of PlayMemories avatar or something like that? In that case, the picture is of the previous owner that somehow survived factory resets?

ma1co, 2020-01-08:
@AlexDaniel Interesting... Could you upload the Backup.bin file to my [dropbox](https://www.dropbox.com/request/WXQ1IKIoPo4ECK8CFxRx) so I can try to find out what this is?

AlexDaniel, 2020-01-08:
@ma1co done.

ma1co, 2020-01-08:
@AlexDaniel The face is stored in backup property 0x00060009 which is used in av-cam.bin from file fc_data_frc.cpp. It seems to be related to face recognition if I'm not mistaken. Maybe it is the last face that was detected or tracked? If you want to play with this, can you try to make it change by trying all kinds of face detection modes? Also try tapping one of the detected faces to go to "face priority mode".