ma1co/OpenMemories-Tweak #10: Button remapping

chungdha, 2016-04-25:
Great job you guys are working on this!

Think the biggest thing and maybe least hardest to reprogram is remap the record button. Either enable the shutter button be used to start recording and stop or C1 button to start and stop recording , like program it so C1 and the Record button switch position, especially be great for any of the A7 users.

tommiehansen, 2016-04-27:
Another thing related to button mapping would be to let each button have two different purposes by separating a push and a double push. That way we instantly have **twice the number of buttons**.

In order for this to work one would need to go beyond just editing the settings file though.
Since i do not understand java i'll write a basic example in javascript:

```
buttonClick = 0;

function doublePress(button, time){
  buttonClick = buttonClick + 1;
  
  // wait 150 ms for another press to occur
  setTimeout(function(){
    press(button, buttonClick);
    buttonClick = 0; // reset buttonClick count
  }, 150);
}

function press(button, buttonClick){
 if(buttonClick ==  1) {
    execFunction(button, 1); // execute the first bind
  }
  else if(buttonClick == 2) {
    execFunction(button, 2); // execute secondary bind
  }
}
```

Some smartphones that have a physical button works this way, they differentiate between push, double push and long push. Some even offers different functionality depending on other variables (most used being that a push on the "home button" when at the "lock screen" triggers a camera app).

Bostwickenator, 2016-04-27:
@tommiehansen You are right in saying that would go beyond editing the settings file. I mean if you want to write an app that does this on the awesome https://github.com/ma1co/OpenMemories-Framework that has been put together. It's not really the scope of a tweak though.

tommiehansen, 2016-04-27:
@Bostwickenator Yeah, i know -- i just couldn't help myself. :)
As i wrote i do not comprehend java and the FW (Framework) is all java, else i would gladly start building apps even though i don't really like apps since they are slow to use and often quite slow to access. I much prefer native and more "low-level" functions.

ma1co, 2016-04-30:
As @Bostwickenator said, you can change the UI with a custom app. This is out of scope for this project.

closed 2016-04-30 by ma1co