ma1co/OpenMemories-Tweak #12: Device error : SSL

Ben4678, 2016-04-25:
Can't install the app on A7S, get following error :

Starting download to camera
Progress: Uploading 0%
Device error: SSL

Camera is switched to MTP mode, connected via USB to a windows computer, using browser IE explorer. I've tried installing the PMCA plugin in Firefox, but it crashes everytime I try to install.

I've tried above steps on 2 different computers now, same result. Any idea what is causing this?

Thanks

ma1co, 2016-04-26:
Set your camera's date & time, that should do it.

jasonhlhung, 2016-04-26:
Same problem here, even on mac. 
What do you mean by set camera's date & time?

ma1co, 2016-04-26:
In your camera menu, go to toolbox -> date/time setup -> date/time and set it to the current date. It's probably a few years off if you see the SSL error.

jasonhlhung, 2016-04-26:
The date of my a7s is correct and up to date, but I am still getting the problem

podmanvn, 2016-04-26:
So u should try to use with Hotspot shield and change Country to Japan ? I meet it before with my Japan-only camera A7M2

ma1co, 2016-04-26:
Can you install apps from https://www.playmemoriescameraapps.com/ ?

jasonhlhung, 2016-04-26:
No

ma1co, 2016-04-26:
Well, then I can't do anything. Blame Sony :)
You can try this instead: https://github.com/ma1co/Sony-PMCA-RE (download the newest release). You have to be familiar with the command line however.

ma1co, 2016-04-26:
the problem could as well be some anitivirus software (or similar) that MitMs your secure connections. Try disabling it (maybe on the official site if you don't trust us :P)

jasonhlhung, 2016-04-26:
Great! 
Successful installation after disabling the antivirus
Million thanks :)

Ben4678, 2016-04-26:
I managed to make it work in the end by re-installing the PMCA plugin, restarting the computer and disabling the antivirus shield (Avast). not sure which one of these steps did it but finally it installed :) 

Great app, thanks!

ma1co, 2016-04-26:
Probably avast :)

closed 2016-04-26 by ma1co

grimmerawr, 2016-06-07:
Hello all - I can confirm I had this exact error until I disabled Avast for 10 minutes and it worked immediately - did not even need to restart the PC. Thanks for the advice!

maki869, 2016-09-02:
Hi
I have the same problem, I get -Device error: SSL
i disabling the antivirus.
The date and time are updated.
i Can install from https://www.playmemoriescameraapps.com/
but But I still get :Device error: SSL
Can someone  help?

Fantazijus, 2018-06-24:
Hi,
I had same problem.
Solved:
Win 7 64bit.
ESET Internet security (uninstalled)
Used Internet explorer 32bit (run as administrator)