ma1co/OpenMemories-Tweak #13: [Feature Request] PAL to NTSC setting change triggers a warning every time camera starts!

sids911, 2016-04-25:
Thanks for this release! I have a Feature Request -when a PAL default camera is changed to NTSC in a region that does not support PAL (and vice versa), the camera triggers a warning message "Running on NTSC" which can be dismissed by half pressing the shutter button.
This nag screen appears every time the camera is restarted or awake from sleep. 

It would be great if this nag screen can be configured NOT to appear at every startup when video format is changed from default. 

Thanks!

Bostwickenator, 2016-04-25:
Already an issue for this

sids911, 2016-04-25:
Agree! Sorry, got a little excited and Trigger happy when I saw this tweak!

mikeytown2, 2016-04-25:
Duplicate of #4?

closed 2016-04-26 by ma1co