ma1co/OpenMemories-Tweak #14: [REQUEST] Enable Remote Video Record for A6000, A7, and A7R

TeamRainless, 2016-04-25:
You can take a photo remotely... you can enter video mode remotely... but for some odd reason it won't let you record remotely on the above-mentioned cameras. You can do it on the A5100 and almost every other Sony camera, but you can't do it on those... for no apparent reason. Please make it so...

henrikhj, 2016-04-26:
As well as the RX1R2.

Boldbrew, 2016-04-26:
I guess it'll be a problem to find a proper configuration file from QX or other standalone cameras with all the necessary properties.

ma1co, 2016-04-30:
This is a problem with the SmartRemote app, since the [MediaRecorder API](https://github.com/ma1co/OpenMemories-Framework/blob/master/stubs/src/main/jasmin/com/sony/scalar/media/MediaRecorder.j) seems to be there since API version 2.3. Try patching the SmartRemote apk directly.

This is out of scope for this project.

closed 2016-04-30 by ma1co