ma1co/OpenMemories-Tweak #2: anamorphic

xcmovie, 2016-04-25:
Hello, is there any possibility to creat 21:9 aspect ratio with true 1080p in video mode?
I just cant use wide angle lens with anamorphic adapter, so I would be very happy if there is  some solution to recording 1080p with different than 16:9.
thanks for info.

ma1co, 2016-04-25:
See the FAQs ("Can you add feature X?"). This is a feature that isn't already there, so probably not. Sry to disappoint...

toferzim, 2017-12-01:
At least it would be greate to have an option to change aspect ratio for the display or for the live view in the phone app. If it records file in 1:1 it is ok, but is it possible to change the aspect for monitoring to 2x anamorphic?