ma1co/OpenMemories-Tweak #3: enable PAL on NTSC only A7's

orico, 2016-04-25:
some A7's are grey import and are limited to NTSC only. could be nice to enable PAL 25\50 on these cams.

Bostwickenator, 2016-04-25:
Inversely many A7s and A7iis are PAL and display warnings on every boot when running in NTSC mode. Maybe we can do something about this. I'm happy to help.

mikeytown2, 2016-04-25:
Duplicate of #4?

Bostwickenator, 2016-04-26:
Different issue to #4 this is about enabling the mode not disabling the warning

ma1co, 2016-04-30:
@orico Where is your camera imported from (i.e. which region setting is it)?

orico, 2016-04-30:
its a friend's camera, gray imported to Israel, not sure where it is originated, will ask.

puredirect, 2016-05-02:
Hi ma1co! I have the same problem, ntsc locked camera, bought from USA. Please help! :)
I didnt know its ntsc only, i live in europe where we use pal.

ma1co, 2016-05-06:
There is a new option in the region tab. Please test and let me know if it works.

puredirect, 2016-05-06:
sorry but im not at home till monday, ill test is as i get home, thank you!

orico, 2016-05-06:
will ask some people.

puredirect, 2016-05-09:
Hi! The camera is now pal ready :D You are a GENIUS. Thank you very mutch!!!!!!!
I test it with several preset and its working fine!

orico, 2016-05-09:
awesome, good job. couldnt get anyone to test it yet.

closed 2016-05-09 by ma1co