ma1co/OpenMemories-Tweak #4: Disable PAL/NTSC warning on startup

ma1co, 2016-04-25:
Seems to be the most requested feature

Bostwickenator, 2016-04-25:
And not just here. https://community.sony.com/t5/Alpha-SLT-DSLR-Cameras/quot-Warning-you-are-in-NTSC-Mode-quot-wtf/td-p/504303

How does one go about discovering a new offset in the settings file? If I can assist with this I'd be more than happy to.

ma1co, 2016-05-06:
There is a new option in the region tab. If you disable the option, the PAL / NTSC selector will be hidden in the menu, but the NTSC warning won't show anymore. If you ever want to switch back to PAL, just enable the option again. Please test and let me know if it works for you.

Bostwickenator, 2016-05-06:
My Hero!

r3dm4n, 2016-05-07:
Works on the A7 mk1
Thank you ma1co !

closed 2016-05-07 by ma1co

jaredcmurphy, 2016-11-26:
Hello,

I am new to GitHub and do not understand how to get this feature on my Sony RX100M3 camera. Do I need to upload this code (and where do I find said code) to the camera somehow? 

Thanks
Jared

briicherri, 2018-03-15:
I cannot seem to find the region tab on the A7S to be able to disable this
any one help?