ma1co/OpenMemories-Tweak #6: ProtectionActivity doesn't work on android 4

ma1co, 2016-04-25:
On non-Japanese cameras an error is shown, but the unlocking succedes. The workaround for Japanese cameras doesn't work.

This is most likely because we can't create /setting (/android is not writable).

sonovice, 2016-04-25:
For future reference:

Title: "NativeException"
Msg: "Backup_sensor_cmd_ID1 failed"

a7rII @ v3.0

ma1co, 2016-04-25:
It should be resolved for now as long as you don't have a Japanese camera. For devices set to Japanese region, this will take some time...

BreezeWindy, 2016-04-25:
Same here, I bought a a7ii in Japan, and the camera menu language is japanese, and the firmware is the lastest ver. 3.10. When I try to activity all language via your app, it shows "Cannot change settings because protection is active. Please disable protection and try again.". So I try disable protection in your app, and it shows "/setting/backup.bin: open failed: ENOENT (No such file or directory)". Therefore, I can't activity other languages. I sucessfully unlock the 30mins video limit via your app, which is wonderful and thank you very much. I hope the language issue can be fixed soon. Thank you again, cheers.

podmanvn, 2016-04-26:
Please fix the language issue for a Japanese-only camera. 
Always shows "/setting/backup.bin: open failed: ENOENT (No such file or directory)". 

Please please please...

tpsl400, 2016-04-26:
Hi Ma1co,

I tried on firmware 3.10 and got the same error:
Could you please help me to fix this? your tool is awesome every-way  :+1: 

[image]

My Device:
•Name: ILCE-7M2
•Product code: **********
•Serial number: ********
•Firmware: 3.10
•Battery: 39 %
•Total storage space: 104.4 MB
•Free storage space: 95.7 MB

Thank you so so much!
Br//

podmanvn, 2016-04-27:
Hi Ma1co, would u please release new version to fix this issue ? 
We really need this Language patch for A7M2 japan-only camera.
Thank :)

ma1co, 2016-04-28:
@sonovice: With the new version, can you enable / disable protection now (i.e. does the display toggle between enabled / disabled when you click the buttons)?

jkairi01, 2016-04-28:
Try to install new version. Still can't disable (same error) but no warning show when press enable.

podmanvn, 2016-04-28:
which version, ma1co ? 
Tried with ver 0.3 but Exception still occurs

ma1co, 2016-04-30:
This issue should be fixed now in v0.4. Can someone please test (just install the app again to update to the new version)?

BreezeWindy, 2016-04-30:
Thank you ma1co, the v0.4 works! Just installed again and tested, at first it showed "navie protection write faiulier" something when I tried to disable the protection. And after that I tried to active the 35 languages but fail. Then I pressed the Menu button to exit and ran the app again. This time the protection was disabled sucessfully without any warning and the 35 languages were actived! Now language can be selected on the menu, I've tried 10 of them and they worked! Thank you again ma1co, you and your app are awesome!

jkairi01, 2016-04-30:
Incredible! It's working for my first try on 0.4. Will looking for bug and report to you. Thank you.

ma1co, 2016-04-30:
@jkairi01 Which camera?

closed 2016-04-30 by ma1co

tpsl400, 2016-04-30:
awesome, you save my life, @ma1co 
Tell me, how can I compensate you :+1: 
Thank you so much much

jkairi01, 2016-05-01:
@ma1co A7RII Japan 3.10