ma1co/OpenMemories-Tweak #15: Video limit didn't seem to work on A7

maxogden, 2016-04-26:
I have an A7 (ILCE-7) and successfully installed OpemMemories - Tweak today. I was on firmware 2.0.0 and tried setting the video length limit to maximum, but after 30 minutes of video my camera stopped recording (e.g. the tweak had no effect). Then I updated to firmware 3.1.0 (the latest) and tried it again, still no effect. 

Thanks for developing this, hopefully this debug info is useful. Happy to try any fixes you suggest.

ma1co, 2016-04-26:
Can you try setting it to 5 seconds?

maxogden, 2016-04-26:
I changed it to 5 seconds and it worked (stopped recording video automatically after 5 seconds). Then I set it back to maximum (13h 1m), and tested again and it stopped recording after 30 minutes (no change).

ma1co, 2016-04-26:
Does it stop exactly at 29:50 / 29:51? Otherwise it could be due to overheating or other factors.

maxogden, 2016-04-26:
Yes exactly at 29:50

[image]

ma1co, 2016-04-26:
Uhm, I can't see the clip length in your screenshot :)

What does the app display for "current limit" if you just open it up?

maxogden, 2016-04-26:
Those are the raw .MTS files it recorded to the SD card, 10:40 + 10:40 + 8:29 = 29:49 (off by 1 due to rounding). The current limit in the app says 13h 01m 00s. At the end of the recording I hear the 'chime' sound it plays when ending recordings + the camera isn't very warm.

ma1co, 2016-04-26:
Sry, my bad. It's a bit weird because it seems to work on the A7M2. I will look into this when I have time.

If anybody else has tested this on the A7, please comment here.

thanhtran207, 2016-04-26:
It doen't work on my sony a7 too :(

ma1co, 2016-04-26:
Do you have the same symptoms? Does setting it to 5s work, but not 30min?

thanhtran207, 2016-04-26:
Im setting it to no limit,  
Well, anyway, thank you very much , for what you did for us
Hope you can bring features on the a7mkii for sony a7 user : 14bit raw and supports phase auto focus with mount adapter 3rd party. ^^
here is my catlog
[TWEAKLOG.TXT](https://github.com/ma1co/OpenMemories-Tweak/files/236468/TWEAKLOG.TXT)

Lummerland1000, 2016-04-26:
It worked with my Alpha 7 rii Camera. Firmware 3.10, japanese import! Great! I had to use windows.IE Browser.Now I have all languages.And the Video function.
Didnt work with Safari and Mac.
Thanks a lot!

kurusutomari, 2016-04-27:
4 Hrs non-stopping recording achieved. No temperature warning at all. Got one 113GB and one 35.2GB files Seems Sony still splits files every 3hrs recording.

ma1co, 2016-04-27:
This issue is about the A7.

chungdha, 2016-04-27:
Could it be that the hack does not work with AVCHD recording and only works on cameras that are recording on XAVC?

chandy227, 2016-04-28:
Also experiencing this issue A7 (Mk 1).  5s setting works, but recording stops at 29:50 when I choose maximum (13h).

ma1co, 2016-04-30:
Everybody: Which settings are you recording in (codec / framerate / etc)?

Chiarascuro, 2016-04-30:
A7r won't record past the 30 min limit even though the tweak says 13h 01m 00s. The 5s test does work. It is set to record in AVCHD,  24p, 1080. Region set to New York/Bogota.  My A7sii and Rx100iii will record past the 30 minute limit.

chandy227, 2016-04-30:
U.S. AVCHD 1080p 60p

maxogden, 2016-04-30:
60p 28M(PS) AVCHD

chandy227, 2016-04-30:
I'll add also 28M (PS) Chicago/Mexico City

r3dm4n, 2016-05-07:
Doesn't work on my A7 mk1 either. 24p 1080 region set to New York/Bogota

fantforever, 2016-05-21:
Have same situation.
5s test break works, but without sound
0:29:50 and 13:01:00 options successfully stores their selection in app even after reboot, but both breaks at 0:29:50 with standard sound

firmware 3.1.0
region is Europe/Moscow
bug occures at 25p/17M and 50p

Thanks for great work!

a7tulp, 2016-05-24:
Same here.
A7 (V3.10) after 29:50 min stops (region is Europe/Paris bug occures at 25p/17M and 50p)
RX100iii no problem >30 min with XAVC-S and AVCHD codec (tried both).

Anyway; thanks for the big efforts so far. Much appreciated by many people fro sure!!

Kedaranatha, 2016-07-07:
confirm. doesn't work on A7r ILCE-7Ra Please HELP  (PS tested on A7ii - no issues. )

details:
**camera**
ILCE-7R
Ver. 3.10

**app**
tweak app version 0.9
java API 2.3
Disable video rec lim : check on (current limi: 13h 01m 00s)

**result**
camera makes beep sound and stops on 29:50 all the time
camera doesn't post any warnings etc

[TWEAKLOG.TXT](https://github.com/ma1co/OpenMemories-Tweak/files/351516/TWEAKLOG.TXT)

C-hiba, 2016-07-19:
confirm:

cameras:
ILCE-7
Ver. 3.10
ILCE-7
Ver. 2.00

app:
tweak app version 0.7.1

same situation:
5s test break works, without sound
0:29:50 and 13:01:00 options successfully stores their selection in app even after reboot, but both breaks at 0:29:50 with standard sound

Sion-Jones, 2016-07-29:
Confirming the issue here as well.

ILCE-7
Ver 3.10

Same as others have reported, the setting appears to take effect, but camera still stops recording video just short of 30 minutes.

arturjaracz, 2016-07-29:
I tested on A5000, A6000, and A7mk1 A7Rmk1.
In all these cameras the same problem - stops at 29:50.
A6000 tested on the new software with XAVCS. Although it stops at 29:50.

friendlygiraffe, 2016-07-30:
Installed on Sony A7 with 2.0 firmware.
Enabled the longer recording length in app settings, said about 13 hours available
Unfortunately all video recordings stop at exactly 29:50.36
This is using AVCHD 1920x1080 and plenty of SD space
Updated to 3.20 Firmware, but still no joy.

Model: ILCE-7
Backup region: CX77801
Tweak app version: 0.9

QuantMcDonald, 2016-09-15:
Same thing for me, the video stops at 29:50 even after setting up the Unlock.
Would love the hack to work of this first version of the Sony A7.

Camera : ILCE-7
Firmware : Ver 3.20
Region :  Europe/Paris
Video Settings : AVCHD / 1080p / 60p 28M (PS)

Merci for your work ;)

red-scott, 2017-02-12:
Just installed and ran a test recording, stops at 29 minutes, 50 seconds.

Camera: ILCE-7
Firmware : Ver 3.2
Region : United States
Video Settings: AVCHD / 1080 24p

[INFO][InfoAdapter] Model: ILCE-7
[INFO][InfoAdapter] Backup region: CX77800_UC2
[INFO][InfoAdapter] Tweak app version: 0.9
[INFO][InfoAdapter] Java API version: 2.3

Included TWEAKLOG

Parts that look interesting:
[INFO][BaseProperty.exists] backup property 0x3c04b6 does not exist
and
[INFO][BaseProperty.setValue] backup property 0x10d008f is protected
[INFO][guessProtected] cannot write backup property 0x10d008f (probably protection)

[TWEAKLOG.TXT](https://github.com/ma1co/OpenMemories-Tweak/files/769561/TWEAKLOG.TXT)

roryphd, 2017-03-08:
So are we still waiting for a fix on this A7 not working? I'm having the same problem. Excited to see this fixed so my A7 can be used for longer videos.

milkagb, 2017-03-18:
On Alpha 7S2 same issue but with 35 min, so strangely it get through the 29'50'' but for only 5'10'' more then stop.....
Hope you will able to fix it and thanks again for the achieved work.

Corstiaan, 2017-08-04:
Same problem here :-(
ILCE-7r, firmware 3.2
Installed tweaks successful, setting limit to 13h but recording still stops at 29'50"

milkagb, 2017-11-25:
@ma1co is using pmca-gui v0.15 on Firmware 4.0 is working finally? I have now a 7M2 and would like to test to unlock the recording time limit also, maybe it will work on this one. Thanks!

ma1co, 2017-11-25:
@milkagb pmca-gui should work since [June 8](https://github.com/ma1co/Sony-PMCA-RE/commit/f0ceb822e75d190279daab5c492d64b5973264ca) (the day FW 4.0 was released). If you have any problems, please let us know.

gimdongwoo, 2018-02-22:
Any update?
It is still not work...
A7 mark1's last firmware is 3.20.

ShatteredPixelz, 2018-05-10:
Did anyone figure out why this happens? Or is this one mystery that can't be solved?

Jeziorko, 2018-10-27:
I have this same problem. Do you have any solutions?

ShatteredPixelz, 2018-10-31:
none at the moment...

Jeziorko, 2018-10-31:
;(

reugelo, 2018-12-08:
Hello and thanks for your work

Same problem here. 
I do not hsve the option for 5" then I can try.
Camera: ILCE-7
Firmware: 3.2
Installed OpenMemories Tweak
Changed limit in app to Disable Video limit 13h 01m 00s
Recording video at 1080 50p AVHCD stops at 29' 50"
Recording video at 1080 25p AVHCD stops at 29' 50"
Hope we can solve it

na7681, 2019-06-09:
I have the same problem. Any news. Thanks

Jeziorko, 2019-12-02:
Any news or Any solutions?

acehobojoe, 2020-03-10:
Same issue here on my A7

tomekdersuaaron, 2020-05-15:
Same issue here:
Camera: ILCE-7
Firmware: 3.20
OpenMemories Tweak: 0.11

Recording video at 1080 24p AVHCD stops at 29' 50"

Cosmo, 2020-06-28:
Same on my ILCE-7 :(

seifer7, 2021-06-27:
Same issue here.
Camera: ILCE-7
Firmware: 3.20
OpenMemories Tweak: 0.11

Tweak app says recording limit is set to 13h 01m 00s
Video actually stopped at 29:50

Wish I'd read these issues earlier. I tried to record an event (luckily it wasn't the only camera angle) and didn't know it stopped :(

Has there been any progress on this issue or hope of it being fixed?

Capestrano, 2021-06-27:
How are you ?

pete176, 2022-05-25:
In the meantime, has a solution been found for the ILCE 7 camera? We have the same problem: the recording stops just before 30 minutes and the tweak app has no effect.

lawonga, 2023-01-17:
Same issue here ☹️

KCS74, 2023-02-05:
Same here with A6000.
Pls. can you asist?

jonlewisorg, 2024-01-27:
I’m having a similar issue.  Camera info:
ILCE-7RM2 (a7 rii)
Tweak version 0.11
Android 4.1.2 (SDK 16)
Java API 2.14

I’m a little confused by some of the comments here (about setting an arbitrary recording limit). In the tweak app, under video, the only options are checkboxes for Disable video recording limit (13h 01m 00s) and Disable 4K video recording limit (RX100M4 only)(not available)

I just tried my first long recording since installing the app, and it stopped at 35min (longer than the old limit, but not by much). This was MP4, 1920x1080 30p.