ma1co/OpenMemories-Tweak #16: Error ENOENT, open failed Backup.bin on Sony A7M2 fw 2.0 and 3.10

podmanvn, 2016-04-26:
[image]
I got this error when using "Disable protection" on my A7M2 camera fw 2.0. It couldn't find Backup.bin 
Could u please help me ?
P/S: I can't find fw 2.10 for this camera, can u help?

ma1co, 2016-04-26:
Duplicate of #6. I assume that you have a Japanese-only camera?
No need another firmware, it's just an issue that has to be resolved, it should work on firmware 3.10 afterwards.

closed 2016-04-26 by ma1co

tpsl400, 2016-04-26:
Hi Ma1co,

I tried on firmware 3.10 and got the same error:
Could you please help me to fix this?

[image]

My Device:
•Name: ILCE-7M2
•Product code: **********
•Serial number: ********
•Firmware: 3.10
•Battery: 39 %
•Total storage space: 104.4 MB
•Free storage space: 95.7 MB

Thank you so so much!
Br//