ma1co/OpenMemories-Tweak #17: Unable to compile with Android Studio 2.0

Boldbrew, 2016-04-26:
Could you give any hint how to compile the project with Android studio 2.0 and latest gradle plug-in?

ma1co, 2016-04-26:
You have to unpack the firmware and add `-Lpath/to/nflasha16/lib` to `ldFlags` in *app/build.gradle* (I have to document that somewhere). Otherwise, it should work fine with Android studio (except for jni code highlighting, which is broken atm).

Boldbrew, 2016-04-26:
Thanks for the quick response. I'm probably doing something very wrong way. 2.0 return me the following after i've added ldFlags.add("-Llocal:\\windows\\path")
Gradle 'OpenMemories-Tweak-master' project refresh failed
Error:For input string: ""

ma1co, 2016-04-26:
Escape the backslashes?

Boldbrew, 2016-04-26:
Yes. But it looks like it's not possible to do anything on Windows due to luck of support of 2.3.3 :(

ma1co, 2016-04-26:
It works fine for me on windows. Can you run `gradlew build` outside of android studio (you have to have the gradle wrapper somewhere in your path)?

Boldbrew, 2016-04-26:
> It works fine for me on windows.
I'm surprised. I could not download api 10 platform. I mentioned that it's incompatible with windows

Root folder returns the following for me: http://pastebin.com/Wdjam2HT

Boldbrew, 2016-04-26:
I've got the following http://pastebin.com/Wdjam2HT

Boldbrew, 2016-04-26:
I found one problem for me. I've downloaded zip but need to clone repository instead. With the cloned repository it returns me already known 10 platform problem
FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':app'.
> Exception thrown while executing model rule: BaseComponentModelPlugin.Rules#cr
eateAndroidComponents
   > failed to find target with hash string 'android-10' in: C:\Android\sdk2

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug
option to get more log output.

BUILD FAILED

Total time: 4.753 secs

ma1co, 2016-04-26:
Then install platform 10 with the SDK manager. Oh, and you'll need the NDK as well.

Boldbrew, 2016-04-26:
I'd like but sdk manager say it's incompatible with windows 
[image]

ma1co, 2016-04-26:
Use this answer or simply increase the compileSdkVersion (this should work as long as you don't use any new features) http://stackoverflow.com/a/36392576

Boldbrew, 2016-04-26:
Thanks! I'm able to build it from command line. Unfortunately it's not possible to emulate the camera but it's very different option. Hopefully this issue will help not only to me.

closed 2016-04-26 by Boldbrew