ma1co/OpenMemories-Tweak #20: A7/A7r/A7ii : set minimum shutter speed

ChrisFrance, 2016-04-26:
Hello Ma1co !
First of all I want to congratulate you for your work!
Do you think it is possible to add the setting of the minimum shutter speed shutter on the A7 and A7ii sony? (exists on the A7rii)

thanks in advance !
Chris

lh8511, 2016-04-26:
This already an custom app for that: http://psafs.pixub.com/#..
It is not a permanent solution, however since the camera restores to default values once power off or enters in" sleep mode"

ChrisFrance, 2016-04-26:
ok great lh8511 ! if Ma1co could make permanent solution will be great!
thank you for sharing lh8511, I did not know !
Chris (french user)

entracte1, 2016-07-14:
+1 for minimum shutter speed with auto iso for the Sony A7 !

ngasset, 2016-07-26:
Hi,
does anybody managed to get in http://psafs.pixub.com/# lately.
The page does not seem to exist anymore! 
Could the apk be retrieved from elsewhere ?

NetrunnerAT, 2017-10-14:
Bracket
The bracket application allows to make normal bracketing without big limitations of the camera or other Application available:
It allows more than 3 shots (up to 15)
It have a start timer. So perfect to use with Tripod
Adjust the EV instead of keeping it like done by other application
Changes settings so that you don't forget to do it (Set to Manual mode, Disable Steady Shot, Save as RAW, ...)
Save previous settings to revert back when the application closes
Gives you control of the Shutter Speed, Aperture, Focus Mode (MF, DMF), Focus Point and Focus Magnification

Download:

[url]https://web.archive.org/web/20161106100241/http://psafs.pixub.com:80/download.php?app=Bracket&amp;version=1.5[/url] Version 1.5
Fix: Correct one of the Shutter Speed names.

[url]https://web.archive.org/web/20161106100241/http://psafs.pixub.com:80/download.php?app=Bracket&amp;version=1.4[/url] Version 1.4
Add option (via file) to invert the dial funtions between Shutter Speed and Aperture
Add option (via file) to invert the order of bracket (from -/+EV to +/-EV)
Add option (via file) to set the Shutter delay (from the 2 second default)
Fix: Changed code for it to match with ILCE-5100 / ILCE-5000 instead of A5100/A5000. This should be the correct name. 
Fix: No longer try to set the FlexiSpotSize in NEX-5R/NEX-5T/NEX-6 as default settings (the API doesn't exist)

[url]https://web.archive.org/web/20161106100241/http://psafs.pixub.com:80/download.php?app=Bracket&amp;version=1.3[/url] Version 1.3
Adjustments to the way the focus magnification appears.
Added validations to the settings being defined via file.
Added options to load the number of shots and EV spacing via file settings

[url]https://web.archive.org/web/20161106100241/http://psafs.pixub.com:80/download.php?app=Bracket&amp;version=1.2[/url] Version 1.2
Added the possibility to load different profiles. For instance, one that overides ISO and other that leaves it untouched.
Added some changes to the used buttons. Hope to support now a5000/a5100. Untested as I don't have the camera. Send me an e-mail if it still doesn't work.
Removed the setting of Spot size in NEX-5R/NEX-6 as it was not allowing the application to start

[url]https://web.archive.org/web/20161106100241/http://psafs.pixub.com:80/download.php?app=Bracket&amp;version=1.1[/url] Version 1.1
Usage of Bulb mode in order to have a shutter time higher than 30 seconds
Allows the 16:9 Aspect Ratio mode
It can now load default settings via a file

[url]https://web.archive.org/web/20161106100241/http://psafs.pixub.com:80/download.php?app=Bracket&amp;version=1.0[/url] Version 1.0
First Version

Known Bugs:
With NEX-5R (and most likley NEX-6) it seems there is a issue where the actual bracketing doesn't start until user half-press again the shutter button.
To overcome this, you shall:
- Toggle to DMF mode
- Refocus now 
- Manually adjust if needed
- Switch back to Manual Focus
- Start bracketing

MinShutter
The MinShutter application allows to set a minimum shutter in the camera. 
This is usefull if you are using AUTO ISO/Aperture but don't want the shutter speed to drop lower than a given value.

Important Notes to remember:

The min shutter is kept outside of the application. Even across different modes (Manual, Aperture, ..)
You need to go back to the application to clear it. 
Once the camera goes to sleep or is turned off  the defined MinShutter is lost. 

Download:

[url]https://web.archive.org/web/20161106100241/http://psafs.pixub.com:80/download.php?app=MinShutter&amp;version=1.1[/url] Version 1.1
Fix: Removed Debug log
Fix: Correct one of the Shutter Speed names.

[url]https://web.archive.org/web/20161106100241/http://psafs.pixub.com:80/download.php?app=MinShutter&amp;version=1.0[/url] Version 1.0

MemoryBanks (Planned)
MemoryBanks will be similiar to MR of A6000 and newer cameras but not limited to 3 slots. It will allow the user to:
Save current settings to a file
Load settings from stored file
Note: User will be able to rename file and edit it to remove settings that shall not be applied when the file is to be loaded.

Focus Bracket (Planned)
The Focus bracket application will be similar to the normal Bracket. But instead of adjusting shutter speed it will adjust the focus positions
Notes: 
It will have focus step and number of pictures to be taken
This shall not be possible in NEX-5R (The required method doesn't seem to exist). In a6000 and newer cameras it seems possible.

Time lapse (Planned)
The time lapse application will allow to take multiple pictures with an interval between them
<p>Planned to be support:
Interval between Pictures (need to see how much the camera will allow without sleeping)
Number of Pictures to be taken
Pre defined profiles (Intervals, Number of Pictures, Exposure Mode, Aperture, ...)

Script Runner (Planned)
The script runner will make use of the functions develop for the other applications and allows the user to choose, via a file, what he want to do. Like:
Change mode to Aperture Priority
Change Aperture to F8
Mark Loop Start
Take Picture
Wait X seconds
Repeat Loop X times

tenaciousd93, 2018-03-01:
Hello, site seems to be down at the moment :( Does anyone have the APK of last version?
I already tried https://web.archive.org/web/20161106100241/http://psafs.pixub.com:80/download.php?app=Bracket&amp;version=1.5 but it does not work... Thanks!