ma1co/OpenMemories-Tweak #22: Unlock Bulb shooting  on QX1

leeyun00, 2016-04-26:
Hello Sir,

Thanks for great tools for my a6300 camera. :)
I have some question regarding QX1 camera.
It's smallest camera with APS-C sensor and it's very good concept for astrophotography.
But it's shutter speed is limited to 30second maximum.
If it's possible to extend this to 1~5minutes it'll be very good news for amateur astronomers.

Do you have some ideas for this??

Thank you very much.

Yun

ma1co, 2016-04-26:
It is indeed possible to run custom android apps on the QX1. You have to overwrite the SmartRemote app however, otherwise you can't start it (there is no launcher). It shouldn't be too hard to build an app that uses the Sony API to hold the shutter button down for a while.

leeyun00, 2016-04-27:
Wow it's  great news!! 
But unfortunately I don't know about programming, API, C..  
Could you help me for the work? I'm eager to solve this problem so long time.
I have no experience in GitHub and don't know the rules in here.  I'm sorry for bothering you.

Thank you very much.

Yun

ma1co, 2016-04-30:
Get familiar with java and android programming first. Then look at the [framework](https://github.com/ma1co/OpenMemories-Framework).

closed 2016-04-30 by ma1co