ma1co/OpenMemories-Tweak #23: [Request] Higher Bitrate Video (A7S/A7SII/A7RII/A6300)

beefhammer, 2016-04-26:
Hello! I don't know if this is possible right now, but I would love it if you could enable higher bitrate encoding on Sony cameras. Right now the A7(s/r) series / A6000 series max out their video quality at 50mb/s in HD and 100mb/s in 4K. If you could find a way to at least double the mb/s in video mode, that would really elevate the overall video quality coming out of these cameras and open up more options in post. 

If you do manage to unlock this feature, I should probably mention that there might be a slight chance of over-heating the camera, but I don't think it should be a huge issue (maybe see if it's possible to allow two different bitrates to be selectable?)

Thanks!

Bostwickenator, 2016-04-26:
Is there a reason you think there will be a setting the corresponds to a higher bitrate than those exposed to the user or is this a purely aspirational request?

beefhammer, 2016-04-26:
To be honest, I'm not sure. I thought maybe if you could bypass the 30 minute record limit, there might be a way to tamper with the bitrate. 

But I guess I would say this is a largely aspirational request.

Bostwickenator, 2016-04-26:
Ok cool thanks just checking if you could point out somewhere another model was doing a higher bitrate. Thanks

A7SFAN, 2016-04-27:
Although I cannot say this with any official certainty, I have been told by a Sony tech at NAB that Sony often uses the XAVC encoder chip on many of their cameras in the Alpha, Handycam and XDCAM. The chip is capable of Long GOP as well as I frame only at 600Mbp/s. What Sony does is manufacture the chip and one cost, install it in many of their cameras and "turn off" the CODEC (H.264) features that they dont want that camera to have. So, it could be argued that the CODEC chip is capable of FAR more than 100Mbp/s in UHD. Could that safely be increased to 135 or more? 150Mbp/s? A 150Mbp/s would be a HUGE increase, especially when grading SLOG.

dia3olik, 2016-05-04:
an higher bitrate would be a killer feature especially in 4K enabled cameras. but the best thing coupled with higher bitrate would be enabling XAVC 4:2:2 10bits.... instead of 4:2:0 8 bits...

puredirect, 2016-05-05:
All of these cameras can take picture in 14Bit resolution, so no one can say that the bionz x image processor is not capable of higher bitdepth than 8 bit. With the help of magic lanthern, canon can produce 10 bit image sequence, with this feature a7s2 could beat any professional cameras in the market. As far as I know, a7r2 is 42 Mpixel and it can shoot 5fps continuously, (42mpixel in 14 bit). If one makes some calculations, 4k video is only 8mpixel, 42Mpix is more than five times bigger than 8mpixel, so the image processor should be capable of handling 25fps 8mpix 14 bit resolution photos. 10bit is exponentially smaller.

dia3olik, 2016-05-05:
exactly! and it would be insanely great!!

rustygearwheel, 2016-05-09:
i would appreciate more than 50M in 100/120fps 720p mode of my A7s.

puredirect, 2016-05-17:
I forgot about the 8bit dhmi bithdepth limit, if it could be 10bit it would be prety awesome too

proxiper, 2016-08-20:
If it's possible, 10 bit 4:2:2 output will be awesome, even to external recorder!

puredirect, 2016-08-20:
Ma1co, you mentioned some mounths ago, something like there is a chance to play with bitrate, or setting the codec parameters, is there any improvement? Thx

puredirect, 2017-01-18:
Whatsup? :) any news? Thx

LeighRoy20, 2017-01-24:
Higher bitrate in camera would be awesome, is any one working on this? and raw video?

mturkdfeldt91, 2017-12-12:
[image]
These are some entries I found in the a6000. Seems to show it could potentially be enabled up to 150 MBPS.