ma1co/OpenMemories-Tweak #24: [REQUEST] a5100, a6000, select AF System menu

stephen-park, 2016-04-27:
Resently updated  A7m2, (V2.0) 

The AF system menu is opened. 

It Can select Phase AF Mode or  Contrast AF Mode

I Hope use the option at A5100 and A6000

Thank you

mgilden, 2016-04-27:
The mark ii models have a new and different focus system, which enables features and control the older generation didn't have available. For example, the ability to focus with adapted lenses and use phase AF. The way I understand it, this can't just be "enabled" on the older model as the software doesn't exist, and possibly hardware differences prevent it from being ported even if we were at that stage.

ma1co, 2016-04-30:
I don't think this is possible with a simple setting change.

closed 2016-04-30 by ma1co