ma1co/OpenMemories-Tweak #25: [REQUEST] Disable Skin Smoothing Entry

mgilden, 2016-04-27:
Do you think that Skin Smoothing and/or face detection can be disabled on a global level in the camera? I know there are many entries still to explore and figure out, but one of the more annoying functions of the a5100 and a6000 is that it will not allow us to disable the skin smoothing function when using the hdmi out to record externally. 
See more about the issue here:
https://community.sony.com/t5/Join-the-Inspiration/Please-let-us-disable-skin-smoothing-so-it-doesn-t-show-up-in/idi-p/538535

And a demonstration video here:
https://www.youtube.com/watch?v=sadygI26S5A

Sony has refused to address it, finally acknowledging the problem and fixing it in the a6300, however those of us with the previous generation of cameras have no way around it. I wonder, could one of those firmware entries be responsible for turning it off for good?

mgilden, 2016-05-02:
Please let me know what you think and if there's any additional information about this troubling bug you may find useful. 
It currently still effects the A6000 and a5100. With the a7s it appears to happen on some firmwares when face detection is on, but at least it can be turned off. The former apsc models do not have a good workaround at the moment.

dia3olik, 2016-05-04:
Yeah please! This would be great!

TheGrench, 2016-05-06:
Disabling the skin smoothing feature sure face detection is crucially important.  I believe most user would appreciate that.  As a gimbal user the auto face detection feature is wonderful but  because it also comes along with that dreaded face smoothing feature I lose far too much face detail and my subject's faces appears extremely milky.

Inazumaa, 2016-05-12:
+1 to this request!

tormaid, 2016-05-18:
+1 from me as well! I own an a5100 and it's so frustrating that I can't use the HDMI out because of this "feature."

newimp, 2016-05-20:
I can't believe this is still a thing with the A6000 and A5100. The waxy skin thing is awful and ruins the capabilities of an otherwise wonderful camera. 

I was excited about this 30 min removal thing, because honestly, one way to turn it off is to record in camera as well as externally. Before, it would hit the 30 min mark, stop recording internally, and then the skin smoothing would come right back! Now with this app removing the 30 min limit, it can technically record indefinitely without the bug coming on again... that is, until it overheats. :(

The problem is with the A5100 especially, one of the reasons to record externally aside from breaking the 30 min limit is to keep it from overheating every 20-40 min. Sometimes it makes it to the 30 min mark, sometimes not. Sometimes a bit longer, but never longer than that. 

So please, ma1co , if there's any chance of finding a value in there that actually turns off face detection and/or skin smoothing, AND KEEPS IT OFF, it could be an even bigger find than the 30 min limit!

danielawd, 2016-07-14:
Ho yes! please disable the skin smoothing! I use A5100 as a camera for webcasting true HDMI... smoothing is **so bad**!

I "discovered" if you numeric zoom it goes off!... often i do 1.1x for this....

Great work BTW!!!

Thanks!

marines, 2020-04-23:
I guess all the hopes are lost. How do you do 1.1x digital zoom on the kit zoom lens though?

ProjectNextra, 2020-08-17:
Any update on this?

orestesgaolin, 2020-12-07:
> How do you do 1.1x digital zoom on the kit zoom lens though?

I also wonder if this is even possible with kit zoom lens? I don't see a way to change it.

tjanson, 2020-12-07:
> > How do you do 1.1x digital zoom on the kit zoom lens though?
> 
> I also wonder if this is even possible with kit zoom lens? I don't see a way to change it.

What do you mean specifically? Digital zoom is possible, but only after you’re fully zoomed in with the lens’s optical zoom. (Which may be a dealbreaker.)
You can enable ClearImage Zoom under Settings (cogwheel) → 3rd tab → Zoom Setting.

I’m not sure what would happen with a manual zoom lens (whether you can electronically zoom at arbitrary optical zoom levels).

orestesgaolin, 2020-12-07:
Yeah, it seems that it's impossible to avoid this skin smoothing if I have a kit lens (with zoom). That's a bummer.

orestesgaolin, 2020-12-08:
I peeked into the `libObj.so` to look for some clues if it's possible to disable soft skin effect. I wonder if it would be possible to "corrupt" or disable soft skin effect being applied? I couldn't find any references to the config parameters in the code, though.

[image]
[image]

petrnetusil, 2021-03-30:
> I peeked into the `libObj.so` to look for some clues if it's possible to disable soft skin effect. I wonder if it would be possible to "corrupt" or disable soft skin effect being applied? I couldn't find any references to the config parameters in the code, though.
> 
> [image]
> [image]

What about trying what the same approach what the "clean HDMI" guys managed here: https://github.com/ma1co/OpenMemories-Tweak/issues/70#issuecomment-731626365 ? Basically, writing to the memory via telnet and hence disabling it manually?

petrnetusil, 2021-03-30:
@ma1co would you have an idea what piece in memory could turn the skin smoothing off?

sanmai, 2021-03-31:
Sony had the same issue with some of the next models, and they fixed it in a firmware update. It could be possible to get both firmwares, fixed and unfixed, decrypt them as we can do, and find what changed. I had this idea for quite a while, but I don't have time nor resources to do that. I wonder if you guys have the gumption to take this approach.

mischa85, 2021-03-31:
Getting the Backup.bin of a Sony A6300 and comparing it to the Backup.bin of a Sony A5000 actually made me figure out the address of the Clean HDMI output setting. A simular approach might work here.

sanmai, 2021-04-07:
Several other models had the same issue with skin smoothing, that were fixed with a firmware update. [A quote:](https://www.eoshd.com/comments/topic/18689-sony-hdmi-output-and-the-case-of-the-waxy-skin/?tab=comments#comment-123941)

> I just tested this again with an A7S that ships with a different firmware than the one I originally saw the problem on. And it doesn't do it anymore. It appears that firmware v.1.10 no longer has the waxy skin bug.

mungewell, 2021-04-07:
I've been poking around in the FW image, and notice that 'CautionConfing.so' has a LOT of ASCII describing pretty much any parameter that might be stored within the 'backup.bin'... what we need to do is find a way to associate the textual name with the keys and the values that they are set to.

For example, it has this section of strings...
```
cmnViewSettingNodeNoOsdOutputThroughHdmiOn
cmnViewSettingPropertiesNoOsdOutputThroughHdmiOn
cmnViewSettingNodeNoOsdOutputThroughHdmiOff
cmnViewSettingPropertiesNoOsdOutputThroughHdmiOff
cmnViewSettingPropertiesNoOsdOutputThroughHdmiRoot
cmnViewSettingPropertyListNoOsdOutputThroughHdmiRoot
cmnViewSettingPropertyListNoOsdOutputThroughHdmiOn
cmnViewSettingPropertyListNoOsdOutputThroughHdmiOff
```

We know that these are associated with key 0x01070a47, and that 0x01 means 'OSD on' and 0x00 means 'OSD off'. Can we extract those values auto-magically??? _Once we understand that, we can work on all the other string <-> key associations._

This string also appears in one other place...
```
simon@thevoid:~/camera_fw/a5000/firmware.tar_unpacked/0700_part_image/dev/nflasha15_unpacked_unpacked/lib$ find . -name '*.so' -exec python3 ~/SearchBin-github/searchbin.py -t NoOsdOutputThroughHdmi {} \;
Match at offset:          81030        13C86 in  ./viewUnified2.so
Match at offset:         739702        B4976 in  ./CautionConfig.so
Match at offset:         739751        B49A7 in  ./CautionConfig.so
Match at offset:         739794        B49D2 in  ./CautionConfig.so
Match at offset:         739844        B4A04 in  ./CautionConfig.so
Match at offset:         739894        B4A36 in  ./CautionConfig.so
Match at offset:         739947        B4A6B in  ./CautionConfig.so
Match at offset:         740000        B4AA0 in  ./CautionConfig.so
Match at offset:         740051        B4AD3 in  ./CautionConfig.so
Match at offset:         768041        BB829 in  ./CautionConfig.so
Match at offset:         768090        BB85A in  ./CautionConfig.so
Match at offset:         778247        BE007 in  ./CautionConfig.so
```

I also noted that when I was playing around with keys just before 0x01070a47, my camera showed 'No Card' on the screen.... the section of strings just before this has references to memory cards.

Could the strings be in the same order as the keys?

mungewell, 2021-04-09:
Learnt a bit more, I can list the contents with 'nm' and this gives a little more info about what the objects are. Apparently 'B' means... which means that something else might be setting the actual value.
```
           "B"
           "b" The symbol is in the BSS data section.  This section typically contains zero-initialized or
               uninitialized data, although the exact behavior is system dependent.
```

And the naming of the objects look to be important, I'm guessing that **these** are the names of the keys and perhaps sorting them by address (rather than Alphabetically) might hint at the position in `backup.bin`.
```
simon@thevoid:~/camera_fw/a5000/firmware.tar_unpacked/0700_part_image/dev/nflasha15_unpacked_unpacked/lib$ arm-linux-gnueabi-nm -D --defined-only CautionConfig.so | grep cmnViewSettingNodeRoot | sort > CautionConfig_a5000.txt

001b18bc B cmnViewSettingNodeRootDispButton
001b18e4 B cmnViewSettingNodeRoot24pMode
001b190c B cmnViewSettingNodeRootCustomGridLine
001b1934 B cmnViewSettingNodeRootNoOsdOutputThroughHdmi
001b195c B cmnViewSettingNodeRootClockDateTimeSetup
001b1984 B cmnViewSettingNodeRootPlaybackDisplay
001b19ac B cmnViewSettingNodeRootSceneSelection
001b19fc B cmnViewSettingNodeRootSetupDemoMode
```

There are 199 _keys_ in `CautionConfig_a5000.txt` and 236 _keys_ in `CautionConfig_a5100.txt`.
[CautionConfig_a5000.txt](https://github.com/ma1co/OpenMemories-Tweak/files/6282688/CautionConfig_a5000.txt)
[CautionConfig_a5100.txt](https://github.com/ma1co/OpenMemories-Tweak/files/6282690/CautionConfig_a5100.txt)

[image]

mungewell, 2021-04-09:
One would assume that the a6500 would have to most keys, and the fact that they are in the same order further builds proof that this is the order in `backup.bin`.
[CautionConfig_a6500.txt](https://github.com/ma1co/OpenMemories-Tweak/files/6282803/CautionConfig_a6500.txt)

[image]

sanmai, 2021-04-09:
FWIW a5100 _has_ the option to disable skin smoothing, only it doesn't work unless you're recording. Therefore we can assume toggling any known option won't resolve this issue.

orestesgaolin, 2021-04-11:
> FWIW a5100 _has_ the option to disable skin smoothing, only it doesn't work unless you're recording. Therefore we can assume toggling any known option won't resolve this issue.

Do we know if any newer but similar model has this issue resolved? Can we somehow compare the binaries to see what are the differences in software?

mungewell, 2021-04-11:
In order to 'find a delta' it would be most helpful to have 2 versions of FW for the same camera before/after the fix...

In looking at the settings I'm hoping that something can be figured out. For example does setting 'SoftSkin' to off in the menu _really_ turn it off,in terms of the key(s) that it sets?

We have seen that the work around of 'ClearImageZoom' works to prevent the waxy skin of 'SoftSkin' being still functional when turned _off_ - as presumably the camera prioritizes functions vs CPU power. Perhaps we can make another setting to the keys where the image is always "ClearImage" zoomed in 0.1% for example.

Another approach may be to remove/corrupt the library which provides the SoftSkin processing... removing it totally from the camera.

BTW the amount of zoom can be controlled by the SonyRemoteAPI (WiFi control method that a phone would use). There is a Python app which will allow you to control the camera from a PC, and I made a script to set the zoom. 
https://github.com/Bloodevil/sony_camera_api/issues/73

The 'SmartRemote' app needs to be running on the camera, and this limits the available functions of the camera a little though not to the point where it wouldn't be useful as a HDMI attached camera. The app remembers it was running after a power cycle.

The zoom are 'boxed' into different categories, but don't know if you can cause a ClearImage zoom, whilst holding the Optical zoom at a fixed point (ie. not optically zoomed all the way in).

mungewell, 2021-04-11:
Other random thought about functions vs available CPU - reading the manual it states that ClearImage zoom can not be used in combination with other processing, notably Soft Skin and Smile/Face Detect.

If we also follow this logic, what happens if we enable 'Face Detection/Smile Shutter'? _I don't have camera to hand in order to test..._ Does the camera disable Soft Skin whilst it's trying to detect faces? I assume both are fairly CPU intensive.

If you don't want shutter to actually fire you could use 'Registered Faces' and configure it to look for some random celebrity.

thatcomicsguy, 2021-04-11:
> Other random thought about functions vs available CPU - reading the manual it states that ClearImage zoom can not be used in combination with other processing, notably Soft Skin and Smile/Face [...]

I've got a spare a5000 I can donate for your experiments. I love seeing DIY folks demonstrate human mastery over technology.
DM me if you're interested.

Cheers!

sanmai, 2021-04-12:
> Does the camera disable Soft Skin whilst it's trying to detect faces?

It _does_ focus on faces while soft skin effect goes away when you digitally zoom.

mungewell, 2021-04-12:
@thatcomicsguy Thank you for your kind offer, but I do have a a5000 to test on.

I poked around a bit with the `Backup.bin`, manually changing settings on camera and then dumping it to text to find changes.

It seems that changing SoftSkin settings affects 2 duplicate keys, but oddly the second is reset by default to an 'on' value. I wonder whether this actually makes a difference on HDMI (will test tomorrow when I have camera hooked to screen).
```
id=0x01070112, size=0x0001, attr=0x60:
  01 = SoftSkin "Low"
  02 = SoftSkin "Mid"
  03 = SoftSkin "High"
reset data:
  00 = SoftSkin "Off"

id=0x0107088e, size=0x0001, attr=0x60:
  00 = SoftSkin "Off"
  01 = SoftSkin "Low"
  03 = SoftSkin "High"
reset data:
  02 = SoftSkin "Mid"
```

I also found the keys for FaceDetect. 
```
id=0x0107000c, size=0x0001, attr=0x60:
  01 = FaceDetect "Off"
  07 = FaceDetect "Any"
reset data:
  06 = FaceDetect "Registered"

id=0x01070110, size=0x0001, attr=0x60:
  00 = Smile "Slight"
  02 = Smile "Big"
reset data:
  01 = Smile "Normal"
```

orestesgaolin, 2021-04-12:
@mungewell this sounds really exiting. 

From my time spent on that I have a feeling that the most promising way would be to corrupt the soft skin library or replace it with dummy calls. Not sure if that's possible, but it seems that there's no way to prioritize digital zoom over optical one (that's why I purchased fixed focal length lens) in order to disable soft skin. 

It it would be possible to disable by just setting the keys, then it'd be the most surprising bug that Sony could've fixed but clearly denied to.

MarkErik, 2021-04-26:
@mungewell great work uncovering the section with SoftSkin.

Apologies if this has already been answered, but do we know the level (Low, Mid, High) of Skin Smoothing that is applied automatically?

As an a5100 owner trying to capture sharp video over HDMI, it would be great to have some improvement (without having to use the ClearImage Zoom), and I was wondering would a "not-perfect-but-better-than-before" solution be to force the Skin Smoothing to always be Low in the firmware? (that is assuming that it is using High or Mid by default)

sanmai, 2021-05-04:
> It seems that changing SoftSkin settings affects 2 duplicate keys, but oddly the second is reset by default to an 'on' value.

This one looks very promising. What will be the procedure to change the second key?

0xPIT, 2021-06-07:
I'd be happy to put the money I'd need to spend for a fixed focus lens into a bounty pool for creating a usable, persistent workaround to disable the SoftSkin "feature" in my a5000.

mungewell, 2021-06-08:
@0xPIT I'll admit that I have not found time to look at this, for a number of reasons. One issue was recreating the issue clearly.... my beard doesn't help ;-)

Was anyone able to confirm whether setting each/both of the keys noted above will control/affect the level of smoothing?

mungewell, 2021-06-09:
Found something else interesting. There is the `wrapperSoftSkinEffect.so` file, which can be dis-assembled and is has 'get' and 'set' functions. At the end of each of those functions are some numbers which look familiar.

```
simon@thevoid:~/camera_fw/a5000/firmware.tar_unpacked/0700_part_image/dev/nflasha15_unpacked_unpacked/lib$ arm-linux-gnueabi-objdump -d -C wrapperSoftSkinEffect.so |vim -

00001a54 <WrapperSoftSkinEffect::getSoftSkinEffect(ViewBaseProduct*, WrapperSoftSkinEffect::WRAPPER_SOFT_SKIN_EFFECT*, Wra
    1a54:       e92d 4ff0       stmdb   sp!, {r4, r5, r6, r7, r8, r9, sl, fp, lr}
    1a58:       b089            sub     sp, #36 ; 0x24
    1a5a:       f04f 0906       mov.w   r9, #6
    1a5e:       af02            add     r7, sp, #8
[snip]
    1b80:       f7ff eddc       blx     173c <__cxa_end_cleanup@plt>
    1b84:       0a55            lsrs    r5, r2, #9
    1b86:       0107            lsls    r7, r0, #4
    1b88:       088e            lsrs    r6, r1, #2   <----- hey, we know that 0x0107088e is a key!!
    1b8a:       0107            lsls    r7, r0, #4
    1b8c:       047a            lsls    r2, r7, #17
    1b8e:       0000            movs    r0, r0
    1b90:       0470            lsls    r0, r6, #17
    1b92:       0000            movs    r0, r0
    1b94:       046a            lsls    r2, r5, #17
        ...
```

From 'set' reference
```
id=0x01070a55, size=0x0001, attr=0xc0:
  02                                               .
id=0x0107088e, size=0x0001, attr=0x60:
  02
```

From 'get' reference, in addition
```
id=0x01070a54, size=0x0001, attr=0xc0:
  00
```

jokubasver, 2021-12-03:
Any progress? Seems like you guys got quite close to solving the issue

pieceinc, 2022-08-31:
Has anyone managed to figure this out? Just picked one of these cameras up

orestesgaolin, 2022-08-31:
There's still no better solution than a manual lens and increasing the
digital zoom to 1.1x

pieceinc, 2022-08-31:
Yeh ok. Was hoping someone would have worked it out. Ah well. Thanks

OddKMS, 2022-10-05:
Popped by to say I'm in the same situation - picked up one second-hand for streaming use.
Is there any way I can help with this? Either with man-hours or by contributing to a bounty pool?
Figured I'd ask cause it's better than poking around in the dark.

Wabbit007, 2022-10-28:
Progress?

arkadiusz-wieczorek, 2023-02-27:
Has someone tried to compare settings during recording and preview modes? When I start recording, then skin smoothing is gone. Maybe is here something clue in memory?

arkadiusz-wieczorek, 2023-02-27:
Guys, could you point me to where I can find the firmware to disassemble and play with it?

arkadiusz-wieczorek, 2023-02-28:
> It seems that changing SoftSkin settings affects 2 duplicate keys, but oddly the second is reset by default to an 'on' value. I wonder whether this actually makes a difference on HDMI (will test [...]

@mungewell have you tried to change these values to `00`?

arkadiusz-wieczorek, 2023-03-09:
I've just been playing around with these values, I've tried setting them all to `00` but no luck, skin smoothing still exists.

```
/ # bk.elf r 0x01070a54
00,
/ # bk.elf r 0x01070a55
id:01070a55 size:1 data:
02,
/ # bk.elf r 0x0107088e
id:0107088e size:1 data:
02,
/ # bk.elf r 0x01070112
id:01070112 size:1 data:
00,
```

arkadiusz-wieczorek, 2023-03-09:
> There's still no better solution than a manual lens and increasing the digital zoom to 1.1x

I've noticed that I can use a lever to get a digital zoom with value x1.1 but, but monitor displays the progression of this zooming. If you pull the lever very briefly this progression looks intact, if a little longer it looks displaced but still shows x1.1 and the image has a different zooming state. So maybe can we modify zoom stages?

orestesgaolin, 2023-03-19:
When zooming to 1.1x while using manual lens, not only the skin smoothing is disabled but also things like auto exposure or color balance seem to work much worse. For instance some features like spots or lips become more saturated.
I really wish there was a way to keep 1.0x zoom but avoid skin smoothing.

arkadiusz-wieczorek, 2023-03-19:
> When zooming to 1.1x while using manual lens, not only the skin smoothing is disabled but also things like auto exposure or color balance seem to work much worse. For instance some features like [...]

I have to check, usually I use a prime lens from Sigma and I haven't noticed that any problems which you mentioned. Nevertheless when I zoom to 1.1x I see AF-C to center and a face tracking are gone.

sanmai, 2023-10-21:
I wonder if this workaround will work for you folks:

https://github.com/ma1co/OpenMemories-Tweak/issues/70#issuecomment-1763205541

dghodgson, 2024-11-18:
> I wonder if this workaround will work for you folks:
> 
> [ma1co/Sony-PMCA-RE#448](https://github.com/ma1co/Sony-PMCA-RE/issues/448)

Changing the recording format (AVCHD seems to be 03, XAVC seems to be 09) doesn't seem to work here, even setting it to some unused value (e.g. 0a) doesn't work. Smooth skin still persists on the a5100