ma1co/OpenMemories-Tweak #26: [REQ] Really disable RAW compr. when RAW compr. is set to off

tommiehansen, 2016-04-27:
Disable RAW compression when RAW compression is set to 'Uncompressed RAW' for more modes then just singular standard exposure.

This sounds idiotic because of course the RAW compression should be off if the user has set that setting.

However this is not the case since there are several modes that forces RAW compression even though the user has explicitly set the compression to be off aka 'Uncompressed RAW'.

Example of modes that forces RAW compression even when Uncompressed RAW is set:

1. Any continous mode (like continous Exposure Bracketing)
2. Silent Shutter Mode
3. Long Exposure Noise Reduction in combination with a long exposure

dia3olik, 2016-05-04:
yeah especially in silent shooting (AKA e-shutter), it would be great!!!

tommiehansen, 2016-05-13:
I wonder what is possible and not in regards to modifying the settings file though.

benolry, 2016-05-24:
I think that the description of the problem is not entirely correct. When the camera is set to uncompressed raw it writes a 16 bit uncompressed file to the card. In the best case it uses this file to store 14bit information and waste the rest of the file's bits / photo site. In the cases you listed it writes 12bit luminance information / photo site. The timelapse play memories app, for example, disregards the camera settings and has it's own overriding settings that do not offer uncompressed raw as an option.

Compression happens on top of the cases where the camera uses only 12bit if the camera is set up to use compressed raw.

I hope this helps to sort out the differences between compression and effectively used bandwidth.

Ben

tommiehansen, 2016-05-24:
@benolry Yes, that is true -- the file is still _uncompressed_ but 12-bit instead of 14-bit (in a 16-bit container if i remember correctly). The main issue is still the main issue though; "don't fudge my stuff" since the expected behaviour is to get a 14-bit file when using 'uncompressed raw' and not whatever a specific function think is good as a good bitdepth.

Frankly the whole 'Uncompressed RAW' ordeal is terrible. Should ofc have been "Lossless RAW, 16-bit" and when used it should always be Lossless RAW @ 16 bit and not a 12-bit or 14-bit file in a 16-bit container.

benolry, 2016-05-24:
I totally agree with you :) The ideal solution would be a way to switch all modes to 14 bit and at least have only their 14bit saved (along with some exif and preview data). I think cramming 14 bit values into a 16bit container file is called run length compression. This is a relatively easy to implement compression but you still need to have access to some low level api.

tommiehansen, 2016-07-08:
Any sort of update or any new knowledge regarding this?

It would ofc be especially useful for landscapers wanting zero shutter shock using 'Silent Shutter' but is ofc useful in all situations where one want the maximum amount of DR (which one that chooses Uncompressed RAW usually want).