ma1co/OpenMemories-Tweak #27: FTP from camera in A6000

mishimayuki, 2016-04-27:
Is the FTP of images directly from the camera (not going through a smartphone or tablet) a feature that can be unlocked?
thanks

Bostwickenator, 2016-04-27:
Should be quite possible as an app @mishimayuki. But not as a Tweak (i.e. not this project).

@ma1co hey I hope I'm not treading on your toes here. Maybe create another repo for an app index project. People could use that to raise issues against that with app ideas. Or I can do this if it's helpful.

mishimayuki, 2016-04-27:
Thanks Alex.

I'm new of the blog. If you could do it I'd really appreciate it.
Best

ma1co, 2016-04-30:
Yes, you'd have to write a custom app for that.

closed 2016-04-30 by ma1co

obs1dium, 2016-05-02:
@Bostwickenator It'd be nice to have a place to discuss app development.

Bostwickenator, 2016-05-02:
https://github.com/Bostwickenator/OpenMemories-app-index

obs1dium, 2016-05-02:
Ah. You can add http://psafs.pixub.com/ to the list.
Of course it'd be great if @ma1co could link to the repo for increased visibility.

Bostwickenator, 2016-05-02:
Will do. A link would be great. Even better would be a organization to hold
these.