ma1co/OpenMemories-Tweak #29: [Request] Assign Full-Frame/Crop Mode to a custom function button

shinji666, 2016-04-27:
Hi ma1co,

Is it possible to unlock more functions to be assignable to the custom C1/C2/C3/C4 buttons? For instance, Full-Frame Crop mode is a function that a lot of people would like.
I have some software development (not a lot) background and can help contribute if you can give me pointers on where I should start.

Thanks!

MartinNuc, 2016-04-27:
There is a menu option for that. It switches the crop mode based on the lens attached. What is the point to have it on the button? You can always crop during post processing.

Anderz000, 2016-04-27:
The benefit is for video. Being able to quickly swap between them allows for fast run-and-gun changes to FoV without having to change lens, or delve into the menus.

lcorr, 2016-05-01:
It would be great if the crop mode could be saved in the memory recall too. I have all my video settings assigned to the memory recall dial. It's such a pain to still have to go into the settings and change the crop mode every time.

chungdha, 2016-05-06:
But  you can set also Clear Image Zoom to a Custom button and zoom to 1.5x really fast  or even use a wider crop if the lens cover the sensor more or even crop the sensor much more for more zoom.

Mindboxer, 2016-05-07:
You guys don't get it. It's not about cropping the image. It's about being able to switch between Full Frame and APS-C mode quickly and easily, which is really essential to people who shoot video and/or use APS-C lenses. 
It's one of the most requested features for firmware updates.

shinji666, 2016-06-01:
I'm willing to make a donation for this enhancement if this could be done on A7RII, thanks!

chumariesco, 2022-11-30:
This would be an important enhancement indeed, for video recording with fixed lenses APS-C crop allows to quickly zoom in/out.
As @chungdha says, Clear Image Zoom can be assigned to a Custom button and its results are similar in my experience, in the A7SM2 at least.