ma1co/OpenMemories-Tweak #30: [REQ] Enable focus drive on A7ii

MartinNuc, 2016-04-27:
Hi,
I played with your stuff yesterday to 4am! I am really impressed by your framekwork. What a great work.

My first attempts were to make a focus bracketing app for A7ii. Tried to use `camera.shiftFocusPosition(int);` and `startOneShotFocusDrive(int, int)` for that however with no luck.

After some time I found that focus drive support is false on A7ii.

`Camera.Parameters params = camera.getSupportedParameters();` returns: `focus-drive-supported=false`. There is also some `focus-shift-supported=true` but I am not sure what that means.

Maybe its possible to enable focus drive using modification of `Backup`?

ma1co, 2016-04-30:
This is probably not implemented in the java APIs. I don't think we will be able to enable this through the settings...

closed 2016-04-30 by ma1co

MartinNuc, 2016-04-30:
Can anyone check how is it on A7rII?

Bostwickenator, 2016-04-30:
I really doubt there will be a difference there. I can't see a business
reason for hiding that API on one and not the other. As ma1co suggests its
likely just not implemented. You should make a sample app people can
install if you wanna test anyway

obs1dium, 2016-05-07:
On the A5100 `startOneShotFocusDrive` does work, despite `isFocusDriveSupported` returning `false`.
I could not get `shiftFocusPosition` to work, though. [Here's](http://i.imgur.com/N0FYOQw.gif) a sample image sequence I captured with a proof-of-concept app (last frame is the stacked result).

MartinNuc, 2016-05-07:
Interesting. What exact parameters did you use for function call?

obs1dium, 2016-05-07:
`camera.startOneShotFocusDrive(CameraEx.FOCUS_DRIVE_DIRECTION_FAR, 1)`
The second parameter is the drive speed (SmartRemote uses 4, `getMaxFocusDriveSpeed` returns 0 for me). Using large values will throw an exception.
Also note that your `FocusDriveListener` won't necessarily be called after a single invocation with a low drive speed.

Bostwickenator, 2016-05-07:
Nice find

obs1dium, 2016-05-08:
Just published a minimal focus bracketing app [here](https://github.com/obs1dium/FocusBracket).