ma1co/OpenMemories-Tweak #31: Monitor viewfinder

chungdha, 2016-04-27:
Possible to add a this to the quick button or Fn menu list as currently having it on auto it drains the power too fast but always a hassle to go into the menu to change it each time. Or find way to fix it so it stop sensor when you turn off the camera or have a button to switch monitor and evf manually as going into the menu takes too long each time.

Bostwickenator, 2016-04-27:
You can assign this to a button on the A7Rii. I'm not sure how one would implement that by changing the settings. Maybe there is an array of bindable buttons.

MartinNuc, 2016-04-27:
Its possible to change on A7Rii but not on A7ii. I would also appreciate this to be enabled.

chungdha, 2016-05-06:
On Sony A7, A7r and A7s you cannot do this which is quite the annoyance. As its be a great feature as now to save power keep it each time to either just viewfinder or monitor as the problem is, when you turn off the camera the sensor still be using power and causing battery drain while the camera is off.