ma1co/OpenMemories-Tweak #32: Is it possible to unlock the hidden "Application" menu in NEX-6 CN version?

zhm86, 2016-04-28:
The "Application" menu is removed from the NEX-6 sold in China.
No application could be installed nor opened on the camera.
Wondering if there is a flag indicates the hidden status of the "Application" menu.
The Chinese version NEX-6 shares the same firmware with the other versions.

Chinese version:
[image]
Other version:
[image]

ma1co, 2016-04-30:
I'm not sure this is possible, since at the moment, we have to install an app first before we actually change settings.

Can you install apps on your camera over USB? Can you somehow get it to switch to app install mode? Or is the app installation disabled, too?

Arnie97, 2019-04-02:
Well this is a chicken & egg problem. Would it work to flash the firmware from unlimited versions?

closed 2023-08-19 by zhm86

thatmike1, 2026-10-29:
> [!NOTE]
> 🤖 **Claude Opus 5.5 responding on behalf of thatmike1**

Solved on a mainland-China NEX-6 (firmware 1.03, backup region `CX76001_CN2`). Both the hidden Application menu and the refused USB install come from backup settings. Setting them to the values every other market ships with brings both back: Recipe Lab installs and runs from Application List.

**What differs.** Sony's NEX-6 1.03 updater is a single worldwide file, and it carries per-market backup presets (`0110_backup/SYSAVIP-DSLR/EYC_S/CX7600x_<region>.bin` after unpacking with fwtool). Compared with the other 11 presets, CN2 has these turned off:

| backup id | CN2 | all others | effect |
|---|---|---|---|
| `0x010709ed` | `00` | `01` | Main menu. `viewUnified4.so` (function at `0x30d94`) adds the Application category only when this reads 1. It also gates assigning an app to the AEL key or soft key B. Read-only property (attr 1). |
| `0x01640001` | `00` | `01` | USB install. `libInfraScalarExtCmdPlugIn.so` refuses `NotifyScalarDlmode` (the "does not support apps" error, MTP 0x2006) unless this is 1. It's the existing `usbAppInstaller` tweak. |
| `0x00e40002` | `00` | `01` | Set to match as well. Its purpose isn't confirmed. |

**Procedure.** Use Sony-PMCA-RE master from source: `bk patch`, which the read-only flag needs, was added after the v0.18 release. Charge the battery fully first, because the updater refused with "Low battery" at around 50%. Connect in MTP mode and press OK on the camera when it asks to switch into update mode.

```
python pmca-console.py updatershell        # on Windows from source: add -d native if pyusb finds no backend
bk w 01640001 01
bk w 00e40002 01
bk s
bk patch 010709ed 01
bk s
bk r 010709ed                              # should print 01
exit
```

After the restart there are six menu icons, and `pmca-console install -f <apk>` works. I ran the plain writes and the patch in separate updater sessions. Syncing with `bk s` before the patch, as above, should be equivalent.

Undo: run the same writes with `00`. Avoid Setup → Initialize afterwards, because it reloads the CN2 preset and turns all three off again. It's worth keeping a copy of the backup before writing: from Python, `UsbPlatformBackend.getBackupData()` in the updater shell session returns the whole Backup.bin.