ma1co/OpenMemories-Tweak #34: A6300 Rom V 1.00 Time Tweak Not Working

benjaminkungo, 2016-04-28:
Hi Ma1co,

I have installed time tweak and the 5 second test works.  However, when the time limited is set to max, it always ends at 35:08 no matter what memory card, battery life, video type i use.

I tried enabling/disabling/enabling protection, it did not help.

Attached is the log.

Thanks. I hope you can resolve this as its an awesome hack!
[TWEAKLOG.TXT](https://github.com/ma1co/OpenMemories-Tweak/files/239780/TWEAKLOG.TXT)

TeamRainless, 2016-04-29:
Did you update to the latest firmware? (You probably shouldn't do that...)

ma1co, 2016-04-30:
Interesting. I also have reports of people who say that it works for them ([this](http://www.sonyalpharumors.com/a-first-real-sony-hack-removes-the-30min-video-recording-limit-and-unlocks-the-language-menu/#comment-2641794866) for example). Which settings are you trying to record in (codec, framerate, etc). Which region is your camera set to?

benjaminkungo, 2016-04-30:
Sorry ma1co this is a user error on my end.  I tried recording via i720 and it recorded up to 2 hours.  So maybe it doesn't work for all video formats.