ma1co/OpenMemories-Tweak #35: Request NEX7

eruan99, 2016-04-28:
Hi,

Would you have any news on whether or not this tool will be compatible with the NEX-7 cameras? Personally, just being able to change language would be fantastic!

ma1co, 2016-04-30:
The NEX-7 doesn't support apps, so this method won't work. Read the [FAQs](https://github.com/ma1co/OpenMemories-Tweak#i-have-an-old-camera-that-doesnt-support-apps).

closed 2016-04-30 by ma1co

ma1co, 2017-10-30:
@eruan99
There is a new experimental way to tweak settings on older cameras using the firmware update functionality. Read more about it [here](https://github.com/ma1co/Sony-PMCA-RE#tweaking-camera-settings). Keep in mind that you are probably the first person to test this on a NEX-7, so your mileage may vary. If you attempt anything, please let us know!