ma1co/OpenMemories-Tweak #38: Can I use this app on my Sony RX100m2 ?

gordianknot11, 2016-05-01:
Can I use this app on my sony rx100m2 ?﻿

If I use Get camera info:
Loading plugin
Plugin loaded
Starting task
Got task key
Cancelling plugin
Plugin cancelled
Starting download to camera
Error: Network connection

If i use Instal Open Memories Tweak:
Starting task
Got task key
Cancelling plugin
Plugin cancelled
Starting download to camera
Error: Network connection

I tried 2 USB cable with 2 different pc 2 different explorer and 8 different USB port and I have same problem.
I can't see my model in this list:
https://github.com/ma1co/OpenMemories-Framework/blob/master/docs/Cameras.md
I have no hopes?﻿ Can you help me?﻿

ma1co, 2016-05-01:
The RX100M2 doesn't support apps, so you're out of luck. Read the [FAQs](https://github.com/ma1co/OpenMemories-Tweak#i-have-an-old-camera-that-doesnt-support-apps).

closed 2016-05-01 by ma1co

gordianknot11, 2016-05-01:
So there not will be a future development of a modified firmware or app for me?
There are other ways to disable the time limit of 30 minutes on rx100m2?
Thanks

ma1co, 2016-05-01:
Short answer: No, not at the moment. Maybe if we find out how the service mode works.

gordianknot11, 2016-05-01:
ok thanks

ma1co, 2017-10-30:
@gordianknot11 
There is a new experimental way to tweak settings on older cameras using the firmware update functionality. Read more about it [here](https://github.com/ma1co/Sony-PMCA-RE#tweaking-camera-settings). Keep in mind that you are probably the first person to test this on an RX100M2, so your mileage may vary. If you attempt anything, please let us know!

gordianknot11, 2017-11-14:
I have to follow Native Installer procedure?

ma1co, 2017-11-14:
@gordianknot11 First, download and open [pmca-gui](https://github.com/ma1co/Sony-PMCA-RE/releases). Then follow the instructions I linked to:

> In pmca-gui, go to the Tweaks tab and click the Start tweaking button. Follow the directions on your camera screen to reboot to firmware update mode. You can now use the checkboxes to configure your camera's settings. Click Done to reboot back to normal mode.

gordianknot11, 2017-11-14:
Chrome stop download. Why detect a virus?

ma1co, 2017-11-14:
This is because google seems to maintain a whitelist of "safe" downloads and we're not in it. You can get around the warning by opening the downloads list and clicking "Keep dangerous file". See [this](https://github.com/ma1co/Sony-PMCA-RE/issues/51) issue.

If you don't trust the binaries, the source code is available. You just need python to run it.

gordianknot11, 2017-11-16:
If I’m probably the first person to test this on an RX100M2 I can tel you
that work great!
Thank you, let me know how I can leave all feedback you need.

ma1co, 2017-11-16:
That is great news! Which tweaks did you test, which tweaks worked?

gordianknot11, 2017-11-16:
Video limit

This is my log

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Model:              DSC-RX100M2
Product code:       0080814150
Serial number:      02884787
Firmware version:   1.00

Using drivers Windows-MSC, Windows-MTP
Looking for Sony devices

Querying mass storage device
Sony DSC is a camera in mass storage mode

Getting device info
Using firmware for model DSC-RX100M2

Initializing firmware update
Switching to updater mode

Waiting for camera to switch...
Please follow the instructions on the camera screen.
Initializing firmware update
Writing firmware
0%
100%
Success
Success
Done

francisgalvez11, 2021-03-20:
Hi @ma1co, 

I'd like to know if its avaiable on a RX100M2 Japanese version unlock the languages. Idk if @gordianknot11 wrote to you about all the avaiable functions after doing the install.

I'd apreciate ur reply.

s-eremin, 2023-04-20:
I just upgraded the firmware of my DSC-RX100M2 from version 1.00 to version 1.10. Then I connected my camera again using Mass Storage mode, started pmca-gui-v0.18-win.exe, switched to the Tweaks tab and applied the tweak. After that I performed a test recording 90+ minutes of video, and it worked perfectly. Thanks a lot for your efforts, guys! I am so happy I can finally make recordings of longer events. You made my day!

s-eremin, 2023-04-20:
> @gordianknot11 There is a new experimental way to tweak settings on older cameras using the firmware update functionality. Read more about it [...]

I just upgraded the firmware of my DSC-RX100M2 from version 1.00 to version 1.10. Then I connected my camera again using Mass Storage mode, started pmca-gui-v0.18-win.exe, switched to the Tweaks tab and applied the tweak. After that I performed a test recording 90+ minutes of video, and it worked perfectly. Thanks a lot for your efforts, guys! I am so happy I can finally make recordings of longer events. You made my day!

P.S. I tried only removing the 30-minute video recording limit.

Hk1020, 2025-02-25:
Just to let everyone know - I just successfully removed the video recording limit on my Sony RX100M2. I had to use the console version (pmca-console-v0.18), the gui version got stuck after resetting the camera when instructed. This was on Windows 10.