ma1co/OpenMemories-Tweak #39: Faster magnification and remapping APS-C crop button

grimmerawr, 2016-05-02:
Hello, 

Thanks for this fantastic hack - I often have to shoot long events and performances so this will be invaluable.

Anyway, if possible, I have two humble requests for future changes:

Firstly, when pressing the focus-magnification button, a box appears showing the area that will be magnified and then you have to press the button again to actually magnify. I use manual lenses exclusively so having this extra step is really annoying and slows down my work, especially in event situation when I need to focus faster. Would it be possible to remove this step using the hack you've discovered?

Secondly, is it possible to allow re-mapping of the APS-C crop feature to one of the custom buttons or at least to the function menu? This is another feature I use frequently so having to go into the menu, and find the option is annoying and slows down filming.

I'm not sure if either of these are possible using this hack method but either way, thanks so much for listening!

Gauban, 2016-05-03:
Hi, I created an Githhub account just to say, @grimmerawr really hit the nail on the head with this request. I too am in the same situation and those 2 features would be super great to have. 
Thank you for all the work on this.

cheers,

Mindboxer, 2016-05-07:
Agreed. Both these features would be useful. Especially the FF/APS-C toggle assignable ot custom button

r3dm4n, 2016-05-07:
@grimmerawr I'd really love to have this as well .. We should be able to magnify directly when pressing the button, without that box appearing the first time. 
Secondly, why isn't there a custom button setting for switching between FF and APSC anyway ?

shinji666, 2016-06-01:
Very looking forward to these two enhancements!

shinji666, 2016-06-14:
@ma1co I started doing some self studying on IDA and reverse engineering. I used to know assembly language but it has been a very long time since I last used it. But before I dump a lot of time into this, based on what you know so far, do you know if this is something can be done (mostly focusing on the second feature - re-mapping of the APS-C crop mode to a custom button)? Any pointers on where to start will be much appreciated. Thanks!

RS-MI-SO, 2016-06-16:
Hi I use my A7M2 only with Manual focus lens and I´m dreaming of something like permanent Focus-magnification box mode with individual setting menu. Anyway big thanks to **ma1co**

RS-MI-SO, 2016-06-16:
__________________________
|                                                   |
|                                                   |
|                                                   |
|                         X                        |
|______  .                                     |
|            |                                       |
|______|___________________|

chumariesco, 2021-01-13:
I own a Sony A7s II and the lack of a custom button for switching FF/APC-S crop mode is the most obstrusive lack I have, specially when filming spontaneuosly. Many of us use that crop as a digital zoom with fixed lenses. Is there any hope here for a 'hack' or trick in that direction? Other buying a A7s iii which allows it.

Respects.