ma1co/OpenMemories-Tweak #40: Unlock XAVC-S and slog2 on Sony A7/A7R

LXR85, 2016-05-02:
As you might know these two cameras have been left out when almost every DSLM in Sony's Portfolio received the XAVC-S codec and some of them even the SLOG-2. Can you check if there is hidden setting for that? I think it would also fix the 29:50 Minute bug.

TeamRainless, 2016-05-02:
Wait a sec... You mean the A7/A7R *can't* record XAVC-S?  WHAT A BUMMER!  There's like absolutely no reason for it.  They're using the same Bionz processor as the a6000 and all those other cameras.  This is just like the weird thing where the a6000 doesn't save copyright info.  I'm going to have to get digging on this and see if I can unlock those things...

LXR85, 2016-05-02:
Yeah, it is crazy. Sony just wants to segregate their portfolio. There has been a lot of petitions, but you know how Sony treats customers once they bought the stuff: http://community.sony.com/t5/Join-the-Inspiration/A7-A7R-codec-update-for-XAVC-S-and-or-S-LOG2/idi-p/445982

TeamRainless, 2016-05-02:
Precisely why I didn't buy the a6300. If they put a touchscreen... or at least a flip screen... or AT LEAST in-body stabilization on the a6400... then I'll buy that. Until then I'm not going to be bedazzled by their "18k downsampled to 4k" or whatever gimmick they throw out next year.

henrikhj, 2016-05-03:
SLOG-2 on the RX1R2 would also be great!

r3dm4n, 2016-05-07:
XAVC-S on the original A7/A7R would be ...the best thing ever.
S-log2 would be a nice bonus, but I would be extremely with just the higher bitrate of the XAVC-S

LXR85, 2016-05-10:
Here you can see what the slog2 update is capable of: https://www.youtube.com/watch?v=xjrg6RIcSsg

chandy227, 2016-05-11:
If unlocking this codec on the A7 is possible that would be so fantastic.  It blows my mind that Sony hasn't done it.  It's really the only thing I wish my camera did that it does not do.

andywendy, 2016-05-14:
Please add XAVC-S on A7R, that's what i'm missing most. Thanks

entracte1, 2016-05-30:
If you can add XAVC-S and SLOG2 on the A7 it will be really great ! 
Thanks a lot !

henrikhj, 2016-05-30:
I'm willing to make a donation to @ma1co towards this goal (on RX1RII), if a paypal or the like can be setup.

LXR85, 2016-06-01:
> I'm willing to make a donation to @ma1co towards this goal (on RX1RII), if a paypal or the like can be setup.

Me too!

entracte1, 2016-06-01:
+1 for the A7

chandy227, 2016-06-13:
+1 for chipping in a donation toward this goal

r3dm4n, 2016-06-13:
With this level of knowledge, he most probably isn't doing it so we can donate a few bucks. It's his passion.
Just by adding this on his CV, he could get a job as a programmer pretty much everywhere.

Bostwickenator, 2016-06-13:
Money changing hands brings certain _complications_

ma1co, 2016-06-13:
Correct, I don't want to accept any money right now. I enjoy what I'm doing and I'm learning a ton myself. These tasks are quite hard to achieve, so I can't guarantee success. For now, enjoy what we've achieved so far!

puredirect, 2016-06-13:
Ma1co,what do you think about a survey, you put some goals in your mind, that you could achieve and people could vote :)

ma1co, 2016-06-13:
I think we could just check for the most active github issues (mostly this one, anyway) :P

Problem is, this is a feature that doesn't exist on these cameras yet. And to add something like a codec, we need to understand a lot more about the code first. That is probably the biggest issue right now, understanding how everything plays together in the firmware. There's just too much code running!

For now, the most promising routes are video bitrate & framerate (I haven't made any progress there, but there might be some low hanging fruit). Additionally, I'm still investigating how to trigger the service mode. This would allow us to change settings or maybe even run code on cameras without android. If anybody has a working version of SeusEX (with hasp key and everything), let me know.

Other than that, I really have to work on the android framework, this will allow other developers to create a lot of "usability" applications. @Bostwickenator is already doing a great job :)

chandy227, 2016-06-13:
Thanks for the great explanation.  I've been wondering since this whole Open Memories thing came out just how complicated it is especially working through that Android app layer to unlock firmware type stuff (which includes this and many other requests on here).  Seems fairly involved.  Thanks for the work thus far and good research toward understanding how our cameras work software side!

puredirect, 2016-06-13:
I just read the magic word :D video bitrate!!!!! You will be a god for alpha owners :) thank you, i like your fruits :)

IM95, 2016-06-16:
Getting the 6300 soon, Feel that camera has some hidden gems underneath. Hopefully that camera can crank HFR 240 on short buffer :)

lessbones, 2016-06-26:
Thank you so much for looking into this-- ive been wating with bated breath for these additions from day 1 of owning an a7, knowing that it was fully capable hardware wise-- i have always thought sony was going to add slog and xavc s in the "2.0" update, but they probably pulled the feature at the last minute (i think thr only thing different in that update was startup time, whereas a6000 got all the goods)

LXR85, 2016-06-30:
@ma1co Would it help If we were to dump the firmware of both A7 and A7II, so you could compare them? I could provide the dump for the Sony A7, but unfortunately only Firmware Version 3.10.

ma1co, 2016-07-01:
@LXR85 No, that's pointless. We can simply extract and compare the firmware updates. Firmware dumping is most interesting for cameras without firmware updates available.

LXR85, 2016-08-21:
Is there any news? Maybe a super unstable alpha for testing?

LXR85, 2017-05-19:
Any progress?

GU011, 2017-05-24:
@ma1co Any progress?

pl4mbo, 2023-09-29:
Any Progress? @ma1co

TeamRainless, 2023-09-29:
What? Since 2016??? NOOOOO... 😂😂🤣