ma1co/OpenMemories-Tweak #47: Error when reverting back to original languages on latest version 0.7

op204, 2016-05-08:
Don't know if anyone else has already reported this issue, Searched for it and didn't find any mention of it. 

When attempting to revert back to the original languages by unchecking the Enable All languages setting after it has been previously enabled, I get the following error:

BackupCheckException
Unknown region: CX60901_E32

Sony RX100 M4
Running on 1.30 firmware

puredirect, 2016-05-09:
maybe your firmware was too old, there is a list in readme file of cameras

Compatibility
This app should be compatible with all Sony cameras supporting PlayMemories Camera Apps (PMCA). See here for a full list.

2015-06-10	RX100M4	 firmware 2.14

op204, 2016-05-09:
According to the Sony website v1.30 is the latest version firmware for the RX100M4. It was uploaded to the website on April 26, 2016. So, maybe I'm missing something.

https://esupport.sony.com/US/p/swu-download.pl?mdl=DSCRX100M4&upd_id=10777&os_group_id=6

puredirect, 2016-05-09:
Sorry i just copyed from the lunk maye its wrong

ma1co, 2016-05-09:
It's a known issue, a few regions are missing in the app. I'll fix it soon.

op204, 2016-05-09:
Thank you so much for all your hard work. This app is awesome!

ma1co, 2016-05-10:
Should be fixed in v0.8

op204, 2016-05-17:
You fixed it! It works like a charm on V0.8! No more error when reverting back to original languages setting. Thank you so much for all your hard work!!! :D

closed 2016-05-18 by ma1co