ma1co/OpenMemories-Tweak #48: [REQUEST] Enable PDAF for adapted lenses on A7

alekskrs, 2016-05-09:
I would like to add support to enable phase-detect AF for adapted lenses on A7. The feature was added to A7II since firmware 2.0, so I would expect it could be enabled on A7 since they are using same sensor/HW.

alekskrs, 2016-05-09:
I can see that this feature was already requested for A5100 and A6000 in #24, and it was closed as wontfix. I am just hoping that, Sony shares PDAF related code between A7 and A7II, because of HW similarities between them. Also I noted that recently new firmware versions are released simultaneously for those two cameras.

The setting that we are talking about can be seen here:
http://briansmith.com/sony-a7ii-firmware-update-fw-2-00-adds-pdaf-uncompressed-raw/

caputtski, 2016-06-22:
BIG supporter of this request. Would make a HUGE difference for the usability of the first A7. The hardware already is there, so why not enable it also for third party lenses, same way it is for Sony ones?

zhcmax, 2016-06-23:
Definitely hands up with this enhancement, I think AF hardware-wise A7 and A7MII are identical, and there were rumors in the beginning of the year that Sony will release adapeted lens PDAF supported firmware, together with 14bit raw support. So I assume there must be some kind of switch hiding behind.

Sincerely hope you can find a way to enable the function! finger crossing!

chichada, 2017-08-09:
Hi, i need open PDAF in a A7 to use the new adapter Techart PRO Leica M – Sony E Autofocus Adapter:

e are not really contended with the performance in A7. The CDAF in A7 does not allow a very good performance at all. We hope Sony will release firmware to open the 3rd party PDAF in A7 system.

jghauser, 2017-10-29:
This would really improve this camera a lot -- would love to see it!

ShatteredPixelz, 2018-03-05:
So, since this has still not been closed, is there any possibility of this becoming a reality?

TriptonPt, 2018-03-16:
This would be the most useful update ever for Sony A7 Mark I users .

xavierbarcelo, 2018-05-15:
If this could be implemented, then a lot would be gained by Sony A7 users!

denna10, 2019-03-18:
This would be greatly appreciated since it would open many new possibilities for us A7 users! Ma1co, do you think there is a possibility that this can be done? Can I contribute in any way?