ma1co/OpenMemories-Tweak #54: Uncompressed RAW in A6000

ArrakisDune, 2016-05-14:
Not seen it elsewere. The only thing that is trully a necessity for me is uncompressed raw, even if the buffer is full faster, or it takes forever to write to the memory card. I do a lot of night /strartrails/ timelapse and the compressed raw option drives me insane, with all the problems it gives to users like myself.

Can it be implemented? If so, can we have it? Thank you in advance!

TeamRainless, 2016-05-18:
I don't think this can be implemented As said on the main page "keep in mind, this app can only unlock features that are already there".  Uncompressed raw is not a hidden feature on any "A" series of camera.  This hack is not the same as Magic Lantern... Magic Lantern added new features to Canon cameras. This hack just unlocks hidden features.

lessbones, 2016-07-08:
The a7ii series has had the ability to do uncompressed raw since firmware 2.0 in Nov 2015-- so it does exist in some form in those firmwares-- given that the a7ii and the a7 have the same processor/sensor shouldn't their firmwares be cross compatible?  I'm not sure if there are any other hardware differences (RAM amounts and such...)

TeamRainless, 2016-07-08:
I think ALL Sony cameras have the same BIONZ processor.  At least every
camera from the a6000 up has it.  Sensors and things change, but I've never
heard of Sony using a different processor...  With Canon you always hear
about Digic 5 and Digic 6 and all that.. .I've never heard about anything
but the same BIONZ for all Sony cameras... (though the amount of RAM
*could* differ.. I kinda doubt it.)

TC5A, 2017-04-22:
An app to enable uncompressed RAW on the A6000 would be great! Hope this can be a reality in the near future!