ma1co/OpenMemories-Tweak #56: How to support ma1co?

distancerunner, 2016-05-25:
How can we support you? How can we get the file with all 15.000 entrys to find the options for you? Is it necessary? I like the results: 30 Min, No NTSC Message. Its a big help for enhance the user expirience.

andywendy, 2016-05-25:
+ 1

shinji666, 2016-06-01:
+1
Very interested to help as well

ma1co, 2016-06-10:
Well, we'd really need some reverse engineers. Get IDA and start digging through the firmware! The settings file is useless on its own since you don't know what each setting changes. So, learn some assembly language!

If you're a Java / Android developer, build a nice timelapse app / similar.

Report any bugs that you may encounter, it's really important to know what works and what doesn't.

It would be cool if someone could test this on a camcorder, since these run android, too.

If you have a camera with firmware 1.00 (no updates installed), i would appreciate if you could dump and share the firmware (some basic linux console experience required, but I'm glad to help).

Other than that, tutorials are always useful to people having trouble installing the app.

henrikhj, 2016-06-22:
@ma1co I have a RX1RII with version 1.00 of its firmware. Will that be of any help, or are you looking for some of the older models?

ma1co, 2016-06-25:
@henrikhj First, can you confirm issue #59? Second, it would be awesome if you could dump the firmware. The procedure is described [here](https://github.com/ma1co/OpenMemories-Tweak/issues/59#issuecomment-225712775). Let me know if you need any help.

byChillout, 2018-12-11:
> Well, we'd really need some reverse engineers. Get IDA and start digging through the firmware! The settings file is useless on its own since you don't know what each setting changes. So, learn some [...]

Hi, i have a cx405 version 1.00 software
https://ibb.co/72YyBng