ma1co/OpenMemories-Tweak #58: Sony QX1 does not have a screen, can't open the app

f2-florian, 2016-05-28:
I was able to install the OpenMemories-Tweak-release-0.8.apk on the Sony QX1, but since there is no screen there is no way to open the app and change the settings. 
Sony limited the QX1 in particular and I hope with some tweaks this very special camera could some options back. (Video mode!!)

ma1co, 2016-06-10:
You're right, the only way I found to actually run an app is to overwrite the preinstalled SmartRemote app (which seems to be started automatically on boot). We'll have to look into this... What features would you like to unlock?

f2-florian, 2016-06-14:
Thanks for your answer, @ma1co!

The features I miss the most for video:
- Removing the 30 minutes limitation.
- For Europeans very important - switching from 30fps to 25fps!

And if possible, using a better video codec?!

For photography: 
- Enabling Manual settings. At the moment, you can either us Aperture priority or Shutter priority. A real manual mode, would be great.

--
To be honest, I am more interest in the video features. The XQ1 is so small, you could use it in places no other camera could be used (for example during concerts!)

f2-florian, 2016-08-12:
Hey @ma1co, is there anything I could do to help getting OpenMemories to work on the QX1?

Cheers!

troop, 2017-01-07:
In theory it should be possible to to change the app that it can accessed over wifi. From what i have seen yet its possible to get access to the httpserver. If that truly works i dont see a problem to change it with an remote app

troop, 2017-01-13:
Im realy interested to get it working. Also when it means that i have to rewrite the remoteapp. Would be great when you can provide informations about wich namespace it use and with wich intent it get started. Greets from freedcam

ma1co, 2017-01-13:
Last time we used Android bug [8219321](http://googlesystem.blogspot.com/2013/07/the-8219321-android-bug.html) to replace the default SmartRemote app. The built-in app is very similar to the one present on other cameras. They all use the same cert for signing.

(Very) brief instructions:
* In your AndroidManifest.xml, set the package to "com.sony.imaging.app.srctrl"
* Name your activity "com.sony.imaging.app.srctrl.SRCtrl"
* Build an unsigned apk. Make sure the only files in the archive are AndroidManifest.xml and classes.dex
* Get SmartRemote.apk from any other camera or the app store
* Use something like [this](https://github.com/Fuzion24/AndroidZipArbitrage) to patch your code into the app:
`java -jar AndroidZipArbitrage.jar --8219321 SmartRemote.apk YourApp.apk`
* Install the resulting apk on your camera. If everything works it, it replaces the default SmartRemote app. (To get rid of it again, you can reset your camera).

Let me know if you need any help.

troop, 2017-01-15:
works perfect. was able to dump the rom.
currently fighting with smali to extend the smart remote to start a service from an external apk. 
big thx for the brief instructions. keep up the great work :dancer:

troop, 2017-01-16:
getting now that with the masterkey. unsigned or with other key signed apks upload fine and the first time it did work. the only difference is, im here on a other pc(this 64bit win10, first time 32bit win7). tryed also different usb ports. thats now a bit confusing.

```
Querying MTP device
Sony Corporation ILCE-QX1 is a camera in app install mode

Using local server
Starting task
Starting communication
Uploading 0%
Uploading 100%
Downloading 100%
Installing 0%
Traceback (most recent call last):
  File "C:\Users\troop\Desktop\opememories\Sony-PMCA-RE\pmca-console.py", line 67, in <module>
    main()
  File "C:\Users\troop\Desktop\opememories\Sony-PMCA-RE\pmca-console.py", line 53, in main
    installCommand(args.server, args.driver, args.apkFile, pkg, args.outFile)
  File "C:\Users\troop\Desktop\opememories\Sony-PMCA-RE\pmca\commands\usb.py", line 192, in installCommand
    installApp(device, host, apkFile, appPackage, outFile)
  File "C:\Users\troop\Desktop\opememories\Sony-PMCA-RE\pmca\commands\usb.py", line 72, in installApp
    raise Exception('Communication error %d: %s' % (result.code, result.message))
Exception: Communication error 100: Error completed
```

ma1co, 2017-01-16:
That's sony english for "internal error". It means that there's something wrong with your apk (usually something with signing). Are you sure you're using the same cert as last time you compiled the app?

f2-florian, 2017-01-16:
Unfortunately, I can't add anything helpful here.
But thanks, @ma1co and @troop for working on this issue again! \o/

troop, 2017-01-16:
i checked the sign on both apks and they are identical. something most go wrong while deodexing it.
cant install clean deodexd apk too.
my approach:

- apply headerfix on odex
- deodex it with smali and backsmali
- push classes.dex to zip and zipalgin it.

have to use that smartremote from a romdump. 
cant download stuff over the market. get each time login failed, but user and password are correct and work in the browser. but that apk should not be deodexed.

troop, 2017-01-16:
im a fool^^
jarsigner: java.lang.SecurityException: SHA1 digest error for classes.dex

troop, 2017-01-18:
Pls someone make that smartremote availible from store. Craweld 6 hours to find that page again with no luck. I promise to backup it this time :D

f2-florian, 2017-01-22:
@troop, is there any progress? What smartremote-app are you looking for?

troop, 2017-01-23:
that one from the offical pmca store. in german "Smart-Fernbedienung". it should be deodexed.
the  problem with the odexed apk is that i cant get injected because the classes.dex is not signed.
different approach would be to resign all apks. but then the rom dump has to get flashed. but dont know if its then boot  🗡

davidbuzz, 2017-05-14:
any further traction on this..?  I have a couple of QX1s Id like to be able to put a custom app onto.   pmca-console and pmca-gui both find my camera  and as a test, I was able to install the 'Tweak' app to the camera no problem, just can't run it or interact with it etc.     I'd like my app to run on boot and simply trigger the shutter every 2 seconds without changing anything else.  :-)

troop, 2017-05-15:
the approach like its done for the other cams will not work for us.

for the settings i injected the remoteapp to start a background httpserver.
but to get interval working the remoteapp need to get extended or completly rewritten.
what other cams can do with using the screen, need we to handel due http request.

but it seems to inject the remoteapp and start from that a own written app is the best approach.
ma1co gave us all stuff we need. 
the stuff i wrote yet is not worth to get published

KillerInk, 2017-08-09:
With installing the last SmartRemoteApp, the Qx1 has full manual mode.
However, Continouse disappears and the reset button does not seem to function anymore.
It also happens that the camera turns off and on. If this happens the Wifi must be switched off and on. Then the camera remains on

Until ContinouseCapture conflicts, I would like to use the original RemoteApp.

On android I can use the shell like this:
`Runtime.getRuntime().exec("dd if=/dev/nflasha of=/android/storage/sdcard0/DUMP.DAT bs=1M");`

How should the string look like, that it works with the shell?
`Shell.execAndroid("dd if=/dev/nflasha of=/android/storage/sdcard0/DUMP.DAT bs=1M");`

Where is the best place to run the snippet?
onResume? onCreate?

Is it allowed to upload the chopped apk to OpenMemoriesStore?

KillerInk, 2017-08-15:
My Qx1 is deceased.
During the apk installation an error occurred

```
Traceback (most recent call last):
  File "<string>", line 66, in <module>
  File "<string>", line 52, in main
  File "pmca\commands\usb.py", line 224, in installCommand
  File "pmca\commands\usb.py", line 70, in installApp
  File "pmca\installer\__init__.py", line 74, in install
  File "pmca\usb\sony.py", line 457, in receive
  File "pmca\usb\sony.py", line 447, in _read
  File "pmca\usb\driver\windows\wpd.py", line 179, in sendReadCo
  File "pmca\usb\driver\windows\wpd.py", line 138, in _readRespo
  File "pmca\usb\driver\windows\wpd.py", line 133, in _send
Exception: MTP SendCommand failed: 0x-7ff8fffe
pmca-console returned -1
```

When I connect it via usb, it charges only the battery. If I then turn it on it does not go in MTP mode. It loads directly my installed apk or whatever it was installed

**If you want to play with it, do not replace the whole SRCtrl with your code.
Use an Intent to start your apk. It looks as if the SRCapp not only control the remote connection**

RIP QX1 :'/

ma1co, 2017-08-15:
@KillerInk I don't know what you did exactly, but I don't think your camera is dead. The SRCtrl app is not responsible for USB communication, that's done outside the Android subsystem. The only thing I can imagine is that your app throws an uncaught exception which crashes the camera.

Have you tried the hardware reset button (number 18 in [this image](http://helpguide.sony.net/ilc/1440/v1/en/contents/image/i0010_63860.png))? This should reset your camera and delete all installed apps.

KillerInk, 2017-08-15:
I dismantled the camera now. And the reset button is canceled. Post later pics. Maybe anyone has a idea to override it

KillerInk, 2017-08-16:
Looks like a flipflop. By pressing the button, the middle pin appears to be connected to the two outer ones. But my knowledge is very limited
[image]
[image]

aargau90, 2017-10-11:
any update on this? 
@KillerInk can you upload your patched remoteapp with manual mode? 
I do not need more than the manual mode for the qx1

KillerInk, 2017-10-12:
@aargau90 you will have manual mode. but all continouse capture stuff is gone. also hw shutter button worked only when a phone is/was connected.
i made no progress after my qx1 died. looks like they stopped selling it in europa and i bought a a6000.

[here it is](https://drive.google.com/file/d/0B62r8-gIQcLJQWJJMEhYQmNpVmc/view?usp=sharing)

f2-florian, 2017-10-12:
What exactly do you mean with 'continuous capture stuff'!?
Does manual mode also mean changing 30 to 25 fps in video mode?

KillerInk, 2017-10-12:
@f2-florian 

> What exactly do you mean with 'continuous capture stuff'!? 

the qx1 have 3 modes to capture a image:
- Single
- Continouse (takes pics till it get stopped, does not show a preview while capture runs)
- Continouse-Speed(same as above, but shows a preview after some images)

with that it loose that two modes.

> Does manual mode also mean changing 30 to 25 fps in video mode?

I did not try it.

aargau90, 2017-10-12:
First: Sorry for my bad english.
Thank you very much for providing the remote apk. i installed it on my qx1 and it works very good. 
manual mode is sufficient enough for me, i use the qx1 only for timelapse with the qDSLRDashboard as Controller.
I have now only two problems: 
- I can not set the image quality to RAW+JPEG (tried over the Sony API)
- in long exposure times (ex. 20") the camera also captures a "dark frame", this doubles the exposure time which is bad for timelapse.

So is it possible to solve this two problems? i think the remote apk is too old and does not support RAW? Can i extract the remote app from a other camera to use this? i have also a sony a6000 and a7

KillerInk, 2017-10-13:
> I can not set the image quality to RAW+JPEG (tried over the Sony API)

Reset the QX1 to default. Start the default SmartRemote, change there the imagequality to RAW+JPEG.
Install the newer Smart Remote and it should shoot in RAW+JPEG.
That newer Smart Remote is not made for the QX1, thats why some stuff is missing, cause on that cameras you have a display and hardware buttons to change it.

> in long exposure times (ex. 20") the camera also captures a "dark frame", this doubles the exposure time which is bad for timelapse.

My QX1 always captured a darkframe. With modding the apk it should be possible to turn it off.
Its called Long Exposure NoiseReduction on sony side.
But i was happy about because i used it mainly to capture stars.

> i think the remote apk is too old and does not support RAW? 

Its not due its age. Its because that apk is not made for the QX1. you have the same functions as when you connected with your phone the the a6000 or a7

> Can i extract the remote app from a other camera to use this? i have also a sony a6000 and a7

yes you can with OpenmemoriesTweak and adb.

Deph0, 2017-11-29:
Hi- was there any progress made towards using OpenMemories-Twerk without screen for this device?
just like OP, i would like to remove 30min limit.

ma1co, 2017-11-29:
@Deph0 You can use the _Tweaks_ tab in pmca-gui as described [here](https://github.com/ma1co/Sony-PMCA-RE#tweaking-camera-settings).

mungewell, 2017-12-05:
You can also use the new 'bk' commands that @ma1co added to 'updatershell' to read/write keys directly, allowing some more experimentation/investigations.

The 'video time limit' keys are known to be:
https://github.com/ma1co/OpenMemories-Platform/blob/master/api/backup.cpp#L239

Novex, 2018-03-12:
> * apply headerfix on odex

In case anyone was wondering what this relates to, the sony apk's use `100` as their odex version which causes an `org.jf.dexlib2.util.DexUtil$UnsupportedFile: Odex version 100 is not supported` error in baksmali.

Running the following code snippet in the `app` and `framework` directories sets the odex version to `036`, which baksmali seems ok with:
```
find *.odex -type f -exec sh -c "printf '036' | dd of={} bs=1 seek=4 count=3 conv=notrunc" \;

f2-florian, 2018-05-18:
I just used the GUI for the very first time and now, with such a powerful tool could anyone **please** add an option to force 25fps in video mode?? Thanks!

mungewell, 2018-05-19:
Personally I can't help implement java code, but you might try setting the following video key (via telnet or pcma-console).
```
Video Mode (confirmed on HX60)
--
id=0x0107009f, size=0x0001, attr=0x60:
0 = 60/50p
1 = 60/50i ?
2 = 30/25
3 = 24 ?
??
5 = 240/200
```

aargau90, 2018-08-22:
does anybody have the original SmartRemote APK from the QX1 or can Dump/Copy it? I can not reset my QX1 to factory (nothing happened when i press the reset button), but i have a newer APK installed without RAW (see Comment a feq posts above).

gabtrek, 2018-11-08:
[com.sony.imaging.app.srctrl-1.txt](https://github.com/ma1co/OpenMemories-Tweak/files/2563541/com.sony.imaging.app.srctrl-1.txt)

I pulled this file from my Alpha6500, just change the extension for apk.

mrsroboto, 2022-04-25:
I know that this is a VERY old Issue/Topic, but I am also looking for a QX-1 SmartRemote APK Dump/Copy. The newer APK features M-Mode which is very nice, but ONLY the original QX-1 SmartRemote APK let you select JPEG or JPEG+RAW inside the Sony API. On every camera with screen the file-format is selected inside the camera menu. Thanks a lot!

mungewell, 2022-04-25:
@mrsroboto I'm not sure whether anyone will have the APK to offer, but as mentioned above in this thread it is possible (at least with other cameras) to use `fwtool.py` to read and set configuration keys on the camera via the USB port. It may be possible to set the camera into shooting "Jpeg + Raw" and have that setting be maintain when the Remote-API app is used.

As a second suggestion; are you 100% sure that it's the APK (and not the phone app) which is limiting the ability to select Raw?

There's another project, which provides an Python library for controlling cameras from a PC. There are calls for reading the capabilities of the camera and selecting settings.... and now I think of it there was a method for authenticating the camera into developer mode which enabled some functionality on the QX-10.

See difference between stock and authenticated here:
https://github.com/mungewell/sony_camera_api/tree/authenticate3/doc

mrsroboto, 2022-04-25:
@mungewell Thanks a lot for your help. You are 110% correct. The "Jpeg + RAW" key could be changed and that would be great. I just don't know what is the correct key name I'm looking for and hope fwtool.py will not brick my camera if something went wrong.

KillerInk, 2022-04-25:
> Reset the QX1 to default. Start the default SmartRemote, change there the imagequality to RAW+JPEG.
> Install the newer Smart Remote and it should shoot in RAW+JPEG.

mrsroboto, 2022-04-25:
@KillerInk Thanks for that, but this is not possible. "Reset the QX1 to default" is impossible, because the Reset-Button is not working anymore after the new Smart Remote.apk installation. I had my QX1 already in RAW+JPEG before installing the newer Smart Remote APK but now it is JPEG only. So your theory that the quality setting will be copied from the default Smart Remote to the newer is sadly wrong. :-(

KillerInk, 2022-04-25:
for me it worked till my reset button broke^^
what i remember about, that the hw buttons where only working after a phone got connected with the new apk.

raw+jpeg do stick because its writting to the bin till you reset it

KillerInk, 2022-04-25:
found here something. that should be the orignal one from qx1
[SmartRemote431.1.spk.apk.zip](https://github.com/ma1co/OpenMemories-Tweak/files/8554148/SmartRemote431.1.spk.apk.zip)
e

mrsroboto, 2022-04-25:
@KillerInk Thanks a lot for your efforts but sadly I now can't connect the camera anymore. "No devices found. Ensure your camera is connected."

mrsroboto, 2022-04-25:
@KillerInk Just managed to install your provided SmartRemote.apk and this one is much better then the one from @gabtrek. No more camera reboots and on/off turning. The quality is still at JPEG and reset button still does not work (if connected to the phones API or otherwise). So I need o find the key to change via fwtool.py ...

KillerInk, 2022-04-26:
or you mod the apk to your needs.
1. for un/packing  and converting the dex to smali i used apktool.
2 .the you can use something like jadx to convert smali to java.
if lucky the class you need is fully translated. if not you have to fix it.
3. then you can create a new android project, drop there the java class with your changes and build it.
4. extract from the apk the dex and convert it to smali. then copy the new smali file to your project from apktool.
5. rebuild the apk with apktool and inject it with AndroidZipArbitrage into the orginal apk.

2. , 3. and 4. are optional. but i find it easier to work with java then with smali

maybe its enough to let the AuthManager.isAuthorized always return true for the dev mode to get back all settings