ma1co/OpenMemories-Tweak #59: Sony RX1R II and RX10M3 error

pjack1, 2016-05-30:
When installed, open the the language menu, scroll down to around Thai language area, the camera will reboot itself. Also, traditional chinese/simplified chinese language are not listed. Is there any way to get them?

ma1co, 2016-06-10:
There might something different in the firmware of these cameras. But I don't own them and there are no firmware updates out yet. Do you have some linux console experience? Could you dump the firmware on these cameras via telnet? Pleas tell me if you're interested.

pjack1, 2016-06-12:
Hi,

I have no experience in Linux console..... though I would like to help you get this corrected... any other way to help?  By the way, Traditional Chinese language is not shown on the language list, is it possible sony have deleted it from the firmware?

Thanks!

Jack

From: ma1co 
Sent: Saturday, June 11, 2016 4:49 AM
To: ma1co/OpenMemories-Tweak 
Cc: pjack1 ; Author 
Subject: Re: [ma1co/OpenMemories-Tweak] Sony RX1R II and RX10M3 error (#59)

There might something different in the firmware of these cameras. But I don't own them and there are no firmware updates out yet. Do you have some linux console experience? Could you dump the firmware on these cameras via telnet? Pleas tell me if you're interested.

ma1co, 2016-06-13:
We'd really need the firmware dumps to debug this. If you want to try to dump it, it shouldn't be too difficult... I'm not asking you to do anything you don't feel comfortable with, I'll just describe the firmware dumping procedure here:

* In the menu, increase your camera's auto power off time to the maximum
* Enable wifi and telnet in the developer tab. The camera should connect to your wifi router (make sure to configure it beforehand) and display its IP address.
* On your computer, connect to the same wifi. Download and start your telnet client (if you're on windows, use [putty](http://www.chiark.greenend.org.uk/~sgtatham/putty/download.html)). Enter your camera's IP address, select "telnet" and hit "open". If everything works out, a black console window with a blinking cursor should open.
* Ensure you have an SD card with some empty space in your camera. Enter the following command in your console window, double check (very important) and hit enter: `dd if=/dev/nflasha of=/android/storage/sdcard0/DUMP.DAT bs=1M` (since these two cameras run android 4). This command will take several minutes to run (it copies about 500MB to the SD card). Wait until the card LED goes off.
* That was it. You can turn off wifi & telnet. You should have a file called DUMP.DAT on your SD card. Please consider that this file not only contains the camera's firmware, but also personal settings like your serial number and wifi password.

Again, only attempt this if you feel comfortable. No worries if you don't want to. Tell me if you need help.

coolcooldool, 2018-12-12:
I have the same problem, and I'm a programer.
I edit the source code in OpenMemories-Tweak/app/src/main/jni/platform/api/tweaks.cpp and backup.cpp, to custom the language menu.
I found this in backup.cpp source code language array,
0:English
1:Japanese
7:Chinese simplified
8:Chinese traditional

Although I open it but I can't set, I think in the firmware, there is no font, so the camera always reboot.
How can fix it?

Thanks!