ma1co/OpenMemories-Tweak #63: Can I use this app on my Sony A37?

diegorodriguesvieira, 2016-06-04:
Hi friend. I've been tried to install this app on my camera but a get the following error:

Loading plugin
Plugin loaded
Starting task
Got task key
Cancelling plugin
Plugin cancelled
Starting download to camera
Error: Network connection

* Sorry by my english, because I'm from Brazil.

Thank you a lot...

emcapone, 2016-06-08:
No, your camera doesn't support PlayMemories Apps on the camera.
https://github.com/ma1co/OpenMemories-Framework/blob/master/docs/Cameras.md

diegorodriguesvieira, 2016-06-08:
Thank you guy! 👍

closed 2016-06-10 by ma1co

ma1co, 2017-10-30:
@diegorodriguesvieira 
There is a new experimental way to tweak settings on older cameras using the firmware update functionality. Read more about it [here](https://github.com/ma1co/Sony-PMCA-RE#tweaking-camera-settings). Keep in mind that you are probably the first person to test this on an A37, so your mileage may vary. If you attempt anything, please let us know!