ma1co/OpenMemories-Tweak #64: [REQUEST] LCD screen brightness in 4k video / slowmotion

alfi666, 2016-06-06:
Hi All,

The biggest flaw of Sony A7S2 and A6300 (and possibly others) cameras in my view is the lack of brightness once you switch to 4K mode or Slowmotion mode.
If the camera is in regular fullHD mode, you can choose "Sunny weather" brightness mode, which works fine, however, once you switch to 4k mode, this option is disabled and the display brightness goes to default, which is useless outside.

If this could be hacked and the high brightness "forced" to stay on during 4k, this would be highly appreciated.

Is there a way how we could help exploring the tweak options in the firmware?

ma1co, 2016-06-10:
Related to #18?

alfi666, 2016-06-10:
Yes, it seems to be the same issue - apparently more cameras are affected.
At this point the only solution seems to be external lcd which is cumbersome and defeats the advantage of small camera

puredirect, 2016-06-10:
Maybe its because the increased 4k energy consumption, the camera probably got unstabe and they solve the problem with this, maybe :)

alfi666, 2016-06-10:
I cant see how lcd brightness can affect overheating - the LEDs are in the screen so they will not heat up the camera... And also I don't believe that the camera would run on the edge with powe consumption - even in 4k recording it takes almost 2hrs to empty the battery, so taking into account the battery size, the current may not be bigger than 500mA
Considering the size of LCD and the LED's it may use, they cannot be stronger than 1W, which would be about 120mA at max... Assuming normally they run at about half power, the increase of 60mA is well within the limit what battery couls supply.
So to summarize, I highly doubt there is any technical reason for this... I think it is just a way for Sony to intentionally fck up their products to protect the "pro" gear.

puredirect, 2016-06-10:
I have a dream, we will find a programer like magic lanthern on canon, who will get the hell out of these cameras:) i think the hardware is good enough for 10 or 12 bit video with a higher bitrate. If 5d mk2 was good for that 5 years before... A7s2 with those big pixels could destroy the whole camera industry...

crirus, 2018-05-22:
No luck in two years, I guess we can safely say we'll shoot blind 4k on sony mirrorless forever :)