ma1co/OpenMemories-Tweak #69: on camera apps backup

RS-MI-SO, 2016-06-15:
It would be nice to have the ability to make a backup on PC or on SD-Card of all installed apps(Free and payed) and reinstalling it .For safety but also for analysis and learning how there work

ma1co, 2016-06-17:
You can download the free apps directly from the market via the [command line tool](https://github.com/ma1co/Sony-PMCA-RE#usage).

For the other apps installed on your camera, I think you should be able to find a [standard android app](https://play.google.com/store/search?q=apk%20backup) which allows you to do that.

closed 2016-06-17 by ma1co

kwkwkwkw, 2016-11-15:
I have tried to install some of the apps from the google play market and seems none of them can extract those installed in-camera apps from Sony Playmemories Store to the SD card.

Not sure it's because of the privilege of the rights to write to the external storage or the path having problem.

Is there any easier and surely can work with the apk backup procedures ?