ma1co/OpenMemories-Tweak #70: Clean HDMI out for a5000

H4ppyDr4gon, 2016-06-16:
With whole videa over 30min it is now possible to stream with the a5000.  But atm you cannot get a clean image without info on screen. Would be great to get rid of that if possible :)

rdweaver, 2017-10-03:
I want this as well!

rdweaver, 2017-10-05:
I installed the demo app on my Sony a5000 and ran the camera view and that gave a clean HDMI out except for the header bar at the top that says camera. Modifying that would possible be a potential solution.

ma1co, 2017-10-08:
@rdweaver Removing the top bar would not be a problem. However, the android screen is only 640x360, but I don't know if this applies to the live view image as well. How is the resolution and framerate of the image on the external screen?

rdweaver, 2017-10-08:
I did not actually check what resolution and framerate it was outputting to the external screen. It would work as it was for my purposes. I will post an update if I do more testing and check that.

rdweaver, 2017-10-09:
I modified the PMCADemo app and removed the title bar on the camera view. This gives clean HDMI output. There are still two potential issues.
#1. I don't think the resolution is 1080. (for myself, this is not a huge deal)
#2. When HDMI output is plugged in, the display on the camera screen goes black. It would be nice to be able to see the image on there as well.

H4ppyDr4gon, 2018-02-27:
Would it be possible to just get the function from the 5100 to not show info on the screen?

Slavpool, 2018-10-15:
I'm curious if anyone has made some progress on this. The a5100 has "Clear HDMI" setting built in, the a5000 has not. If anyone has managed to get a clean camera view through external HDMI on a a5000 please share, it would certainly make many a5000 users happy!

judgemento, 2018-10-29:
I downloaded the PMCADemo App, but the camera title bar was still there, @rdweaver how do I get that to go away? Thanks man!

rdweaver, 2018-10-29:
@judgemento  sorry, I don't have my a5000 anymore, and I wasn't able to find the code I had changed. Here are the steps I took from my memory.
1. Download the source code
2. Edit code to remove title bar
3. Build your own APK
4. Install APK on a5000

ma1co, 2018-10-30:
In [this file](https://github.com/ma1co/PMCADemo/blob/master/app/src/main/AndroidManifest.xml), try replacing
```
<activity android:name=".CameraActivity" android:label="@string/title_activity_camera">
```
with
```
<activity android:name=".CameraActivity" android:label="@string/title_activity_camera" android:theme="@android:style/Theme.Black.NoTitleBar">
```

judgemento, 2018-10-30:
> ```
> <activity android:name=".CameraActivity" android:label="@string/title_activity_camera" android:theme="@android:style/Theme.Black.NoTitleBar">
> ```

Ok I edited, the text with this, but I tried to install with the PMCA-GUI but I can't select it since its not an apk. How do I format it to APK

Slavpool, 2018-11-06:
Hi, I just managed to make it work. I am no programmer and this is the first time I've done something like this so I will try to explain the steps. I've Win 10 so this steps will work for you if you have Win 10. It took me some hours to complete because I had to install everything and I learned by solving 1 problem at a time...
1- Download Android Studio
2- Download Git for windows (during install make sure that you choose option to set Path)
3- In Android Studio, get the PCMADemo though "Check out project from  Version Control" -> "Git"
4- Enter "https://github.com/ma1co/PMCADemo.git" to and then Choose a place where to save a local copy on your PC
5- Open Android Studio then choose "Open an existing Android Studio Project" 
6- Change the Project structure to match SDK and other settings as described in "https://github.com/ma1co/PMCADemo/blob/master/build.gradle", meaning Android Plugin Version should be 2.3.3 and in SDK Manager you need to install SDK Platform 2.3.3 (Gingerbread).
7- Any existing errors? Use google and you will find the solution. Once your changes are made you want to Build APK 
8- Remove the existing PCMADemo App from your camera before uploading the new one.

My new APK is called PMCADemo-debug-0.7-6-g12f34e4-dirty.apk, as I mentioned I don't fully understand what I've been doing but atleast I solved the issue with having the title, thanks to to @ma1co comment above.

The resolution is sadly the same as you have on the LCD screen, even if you run with HDMI cable. However if running the Option "Display" the resolution of the connected HDMI device is shown, in my case my monitor 1920x1080.

I don't know if it would be possible to edit the PCMADemo app further to switch to 1920x1080 when connected to HDMI, or simply forcing it to 1920x1080..

Hope this helps

mcrichards, 2020-03-20:
I'm looking to get a clean HDMI signal out of my a5000 and to a video capture card for use in an OBS stream. I have installed the app and I see it in my applications list, but I'm uncertain as to how OpenMemories-Tweak can be used for this purpose. 

It seems like you all made some progress on this front. Are you able to provide any guidance? 

Much appreciated!

mrboost, 2020-03-29:
Gain any ground on this? ive been editing all kinds of lines trying to figure out how to have a clean display. (ive got the bar gone, but the quality is trash).

cris-cos, 2020-04-07:
> Gain any ground on this? ive been editing all kinds of lines trying to figure out how to have a clean display. (ive got the bar gone, but the quality is trash).

I myself have not managed to get rid of the bar yet, but for the quality:
if you switch to video mode on the camera it clears the output on HDMI out. 
Photo preview on HDMI out is trash, curiously a million times better for video

kalenjohnson, 2020-04-08:
I haven't looked deeply into either solution, the PCMADemo app or the standard HDMI out. I just picked up a used A5000 today.

I'm not sure if the PCMADemo will be able to switch to a higher resolution, it is pretty low resolution so I decided that wasn't going to work for me as a webcam.

HDMI out works pretty well into a capture card, but as we all know it's not a clean HDMI output. After idling for a second though, the only issue is the aperture, iso, and some other data on the bottom of the image.

I haven't tried it in a meeting yet, but it seems to work, is using OBS. Use the capture card/camera HDMI as a source, then oversize it a bit to essentially crop the bottom portion. Then I use OBS-VirtualCam to create another webcam that can be used in Skype and Zoom, probably recording software as well.

https://obsproject.com/forum/resources/obs-virtualcam.539/

TetraSkies, 2020-05-03:
@ma1co I was wondering if it was possible to change to movie mode in the camera mode within the PMCA Demo app, I notice a significant increase in display quality in movie mode but when I switch into camera mode it reverts the quality.

lucaspape, 2020-05-25:
I forked the demo app and removed everything except the camera: https://github.com/lucaspape/PMCA-CleanHDMI (with title bar removed).
The resolution is not great, I am trying to find a way to activate movie mode

NilsConnlaAbbott, 2020-06-04:
Does anyone know how to even connect the camera to your computer? i saw a video on how to use it as a web cam for streaming, and it makes sense but the guy was vague on how to connect it to a software like OBS. Heres the link [https://www.youtube.com/watch?v=wPIk-NG-l1E](url) . I asked him and he said to hook it to the monitor, so i did and got it to display Via HDMI but i have no idea how he got it into a software for recording or streaming. Anyone know how?

sanmai, 2020-06-04:
You'll need an HDMI capture card. [Like this one](https://www.amazon.com/Elgato-Game-Capture-HD60-PlayStation/dp/B01DRWCOGA/), or cheaper variants. Basically these work like another webcam in the system.

NilsConnlaAbbott, 2020-06-04:
Alright, thanks for the reply i appreciate it

jkh13, 2020-06-12:
I have a nex-6 and outputting over HDMI shows a 1080p resolution but with the OSD permanently on the screen. I wonder if there is a setting that can be tweaked to disable the OSD when using the camera normally via HDMI.

moro97400, 2020-09-16:
> > > > No the resolution is probably under 540p... you can see it yourself using the app as is to see the quality.
> > > 
> > > 
> > > this is bad man :/
> > 
> > 
> > lol, this is a hack though. USB 2 camera quality is never going to be HD
> > > > Any progress?
> > > 
> > > 
> > > Basically, get a capture card, there’s a cheap one for $60 that’s just as good at the cam link, plug the camera into it then into the computer. You can disable the icons in the camera settings, except for the iso and F.4 stuff. I’m OBS or whatever you’re using, just crop it out. I think cropping it is a better option, normal camera quality.
> > 
> > 
> > I agree with this, I basically explained the process 2 months ago in this thread
> 
> This is the best option. :)

Hello, I use an ATEM mini switcher and one of my camera is a Sony A5000 if I crop it to hide the icons it crops all of my others camera :/. If someone give me a custom firmware I can pay :) HMU

lucaspape, 2020-09-16:
> Hello, I use an ATEM mini switcher and one of my camera is a Sony A5000 if I crop it to hide the icons it crops all of my others camera :/. If someone give me a custom firmware I can pay :) HMU

I don't think there is any possibility to develop custom firmware on these Sony cameras. The only thing possible is to use the android subsystem and I don't know/couldn't find a way to get full resolution video while you are in this subsystem.

mischa85, 2020-09-30:
For anyone still looking for a solution:

Clean HDMI setting seems to be at **0x01070a47**:

```
/ # bk.elf r 0x01070a47
id:01070a47 size:1 data:
01,
/ # bk.elf w 0x01070a47 00
id:01070a47 size:1 data:00, 
/ # bk.elf r 0x01070a47
id:01070a47 size:1 data:
00,
```

Reboot cam, info should be gone from it's HDMI out.

moro97400, 2020-10-01:
> For anyone still looking for a solution:
> 
> Clean HDMI setting seems to be at **0x01070a47**:
> 
> ```
> / # bk.elf r 0x01070a47
> id:01070a47 size:1 data:
> 01,
> / # bk.elf w 0x01070a47 00
> id:01070a47 size:1 data:00, 
> / # bk.elf r 0x01070a47
> id:01070a47 size:1 data:
> 00,
> ```
> 
> Reboot cam, info should be gone from it's HDMI out. (tested on Sony A5000)

I will test it as soon as I figure it out how to install this tweak. (I'm on MacOS, If someone can help me it will be appreciated, I'am just a camera guy not a dev ;)

mischa85, 2020-10-01:
- On Mac: Install this: https://github.com/ma1co/Sony-PMCA-RE/releases/download/v0.17/pmca-gui-v0.17-osx.dmg
- On Mac: Install App: OpenMemories: Tweak -> Install selected app
- On cam: Developer -> Enable Wifi + Enable Telnet
- On Mac: telnet x.x.x.x (cam Wifi IP)
- On Mac: Enter commands

moro97400, 2020-10-01:
> * On Mac: Install this: https://github.com/ma1co/Sony-PMCA-RE/releases/download/v0.17/pmca-gui-v0.17-osx.dmg
> * On Mac: Install App: OpenMemories: Tweak -> Install selected app
> * On cam: Developer -> Enable Wifi + Enable Telnet
> * On Mac: telnet x.x.x.x (cam Wifi IP)
> * On Mac: Enter commands

On Mac: Install this: https://github.com/ma1co/Sony-PMCA-RE/releases/download/v0.17/pmca-gui-v0.17-osx.dmg
On Mac: Install App: OpenMemories: Tweak -> Install selected app
On cam: Developer -> Enable Wifi + Enable Telnet
This part is OK
But :
On Mac: telnet x.x.x.x (cam Wifi IP)
On Mac: Enter commands
I don't have any telnet menu on pmca-gui v0.17
Is it on the terminal ?

mischa85, 2020-10-01:
> Is it on the terminal ?

Yep!

moro97400, 2020-10-01:
> > Is it on the terminal ?
> 
> Yep!

Can I DM you somewhere please ?

moro97400, 2020-10-01:
> > Is it on the terminal ?
> 
> Yep!

Doesn't work for me :

telnet xxx.1xx.1.1xx
Trying 1xx.1xx.1.1xx...
Connected to 1xx.xxx.1.xxx.
Escape character is '^]'.

BusyBox v1.13.4 (2012-07-17 15:11:04 JST) built-in shell (ash)
Enter 'help' for a list of built-in commands.

/ # bk.elf r 0x01070a47
id:01070a47 size:1 data:
01,
/ # id:01070a47 size:1 data:
sh: id:01070a47: not found
/ # 01,
sh: 01,: not found
/ # / # bk.elf w 0x01070a47 00
sh: /: Permission denied
/ # id:01070a47 size:1 data:00, 
sh: id:01070a47: not found
/ # / # bk.elf r 0x01070a47
sh: /: Permission denied
/ # id:01070a47 size:1 data:
sh: id:01070a47: not found
/ # 00,

mischa85, 2020-10-01:
> Doesn't work for me :
>
> telnet xxx.1xx.1.1xx
> Trying 1xx.1xx.1.1xx...
> Connected to 1xx.xxx.1.xxx.
> Escape character is '^]'.
>
> BusyBox v1.13.4 (2012-07-17 15:11:04 JST) built-in shell (ash)
> Enter 'help' [...]

Almost!
The actual commands are:

'''
bk.elf r 0x01070a47
bk.elf w 0x01070a47 00
bk.elf r 0x01070a47
'''

moro97400, 2020-10-01:
> Almost!
> The actual commands are:
>
> '''
> bk.elf r 0x01070a47
> bk.elf w 0x01070a47 00
> bk.elf r 0x01070a47
> '''
It's working !!! my god thanks you a lot

moro97400, 2020-10-01:
> Almost!
> The actual commands are:
>
> ```
> bk.elf r 0x01070a47
> bk.elf w 0x01070a47 00
> bk.elf r 0x01070a47
> ```

It's working !!! thanks a lot (tested on a5000 too)

bluesnoz, 2020-10-10:
Thanks so much for this thread, I now have clean HDMI on my a5000 ! Yay

Slavpool, 2020-10-11:
> Thanks so much for this thread, I now have clean HDMI on my a5000 ! Yay

Hi, at what resolution? Is it 1080p?

bluesnoz, 2020-10-11:
Yep, output is 1080p

mischa85, 2020-10-11:
It's 1080p50. Although there doesn't seem to be a way to get true 50 fps out of the Sony A5000, even with shutter times as low as 1/100.

mischa85, 2020-10-13:
> Wait so you got a clean HDMI output 1080p50 with the Sony A5000? Is that it?

Yep. Although every frame is displayed twice.

DrGahu, 2020-10-13:
> > Wait so you got a clean HDMI output 1080p50 with the Sony A5000? Is that it?
> 
> Yep. Although every frame is displayed twice.

Ok so in reality you have 1080p25
Is their a Pal vs NTSC mode which could bump this to 1080p30? 

I'm lurking a secondhand a5000 for streaming / video call and casual photo shooting.
Could be the perfect budget cam for streaming if I can get a clean HDMI with reasonable resolution / framerate (720p+ at 30FPS+ would be great).

mischa85, 2020-10-13:
> Ok so in reality you have 1080p25
> Is their a Pal vs NTSC mode which could bump this to 1080p30?

Yes there is.

> I'm lurking a secondhand a5000 for streaming / video call and casual photo shooting.
> Could be the perfect budget cam for streaming if I can get a clean HDMI with reasonable resolution / framerate (720p+ at 30FPS+ would be great).

The firmware menu lets you select between 1080p and 1080i. HDMI resolution gets set at **0x010700b1** (00 00 00 00 = AUTO | 05 00 00 00 = 1080p | 03 00 00 00 1080i). Might have options for 720p as well?

davepee, 2020-10-16:
Is there some guide how it do on Linux?

devfriber, 2020-10-21:
> Thanks so much for this thread, I now have clean HDMI on my a5000 ! Yay

Hello. I have only black screen. Can you help me?

quangvdo, 2020-10-23:
> > Thanks so much for this thread, I now have clean HDMI on my a5000 ! Yay
> 
> Hello. I have only black screen. Can you help me?

Hit the shutter button.

Thank you @mischa85 this is wonderful!

sirbring9, 2020-10-24:
Hi guys, i'm new here and have zero knowledge in programing. I read every post here and managed to install pmca-gui. I installed openmemory on the camera but still don't understand how to get that clean HDMI that you apparently all managed to get. I also installed an apk that someone gave me from the top of the thred but i got a black screen and my camera froze. Can you give me step by step what to do ? I tried Android studio for 3min before uninstalled it because that's a knowledge i don't have. 
Thank you all for your time spent here.

Ascendancer, 2020-11-03:
> * On Mac: Install this: https://github.com/ma1co/Sony-PMCA-RE/releases/download/v0.17/pmca-gui-v0.17-osx.dmg
>
>     * On Mac: Install App: OpenMemories: Tweak -> Install selected app
>
>     * On cam: [...]

You made my day! Thank you so much for saving my butt here <3

Treinkapitein, 2020-11-04:
> 
> 
> I forked the demo app and removed everything except the camera: https://github.com/lucaspape/PMCA-CleanHDMI (with title bar removed).
> The resolution is not great, I am trying to find a way to activate movie mode

Hi, Would you mind sharing the compiled version? I've cloned and compiled the repo, but as soon as I launch the app, my camera freezes.
Thanks in advance!

lucaspape, 2020-11-04:
> Hi, Would you mind sharing the compiled version? I've cloned and compiled the repo, but as soon as I launch the app, my camera freezes.
> Thanks in advance!

It currently doesn't work. I can try fixing it on the weekend but right now I don't have setup to test it

Treinkapitein, 2020-11-04:
Hope I can help some people out:
https://github.com/Treinkapitein/PMCADemo/releases/download/1.0/PMCA_CleanHDMI.apk

Lolloitalia, 2020-11-04:
> Almost!
> The actual commands are:
>
> ```
> bk.elf r 0x01070a47
> bk.elf w 0x01070a47 00
> bk.elf r 0x01070a47
> ```

I don't know you but from Italy I send you a giant thank you. You literally saved my ass after years of testing and never going well. 
Grazie mille Bro, grazie mille.

Lolloitalia, 2020-11-06:
> > > Thanks so much for this thread, I now have clean HDMI on my a5000 ! Yay
> > 
> > 
> > Hello. I have only black screen. Can you help me?
> 
> Hit the shutter button.
> 
> Thank you @mischa85 this is wonderful!

I noticed, however, that the focus frame always remains. do you confirm?

AM555, 2020-11-18:
Is the process the same on a PC as it is on a Mac?

AM555, 2020-11-19:
> 
> 
> I seem to have become unstuck,
> 
> I have managed to install openmemories - tweak, I have enabled the telent client in windows 10, what do I do next?
> 
> When I enable Wifi on the camera do I just connect to the same router as my laptop?
> 
> Thanks in advance!!

I have managed to connect via telnet now. I shall plug it into my capture card and test the HDMI output.

laurismikals, 2020-11-21:
@AM555 

Using Windows 10

- Download pmca-gui - https://github.com/ma1co/Sony-PMCA-RE/releases/download/v0.17/pmca-gui-v0.17-win.exe
- Download PMCA_CleanHDMI - https://github.com/Treinkapitein/PMCADemo/releases/download/1.0/PMCA_CleanHDMI.apk
- Connect Sony a5000 with USB
- Run pmca-gui-v0.17-win.exe that you just downloaded
- When GUI opens press "Get camera info" to make sure that the camera is connected. The window should display camera model and other info
- Go to "Install app" tab
- Make sure that "Select an app from the app list" is ticked
- From the dropdown chose "OpenMemories: Tweak"
- Press "Install selected app"
- Tick "Select an apk"
- Press "Open apk..."
- Find "PMCA_CleanHDMI.apk" that you just downloaded
- Press "Install selected app"
- Disconnect the camera from the PC
- In camera go to Menu -> Application List -> OpenMemories: Tweak -> Developer
- Tick "Enable Telnet"
- Press on "Wifi settings"
- Press on "Access Point Set"
- Choose your wifi and enter it's password
- Go back to OpenMemories: Tweak -> Developer. Tick "Enable Wifi"
- Wait till it connects. It will display IP address which you will need later.
- In your PC open Settings -> search "telnet" -> click on "Turn Windows features on or off" - Select "Telnet Client" -> Press “OK"
- Open terminal. To do that type "terminal" in Search -> click on "Windows terminal"
- type "telnet <put the IP address here>". For example "telnet 192.168.2.9". Press "Enter".
- Now copy these commands in terminal:
```
bk.elf r 0x01070a47
bk.elf w 0x01070a47 00
bk.elf r 0x01070a47
```
- Press "Enter".
- Take your camera. Select "Movie" mode
- Insert HDMI cable. Now your Sony a5000 should have **clean HDMI out** with 1080p resolution!

Thanks to the community! This is great! 😄

brettvachon, 2020-11-22:
> I noticed, however, that the focus frame always remains. do you confirm?

Try this: Under Camera Settings, Third Tab, set “Focus Area" to Wide. The box should disappear.

alexander-manariotis, 2020-11-27:
> @AM555
> 
> Using Windows 10
> 
> * Download pmca-gui - https://github.com/ma1co/Sony-PMCA-RE/releases/download/v0.17/pmca-gui-v0.17-win.exe
> * Download PMCA_CleanHDMI - https://github.com/Treinkapitein/PMCADemo/releases/download/1.0/PMCA_CleanHDMI.apk
> * Connect Sony a5000 with USB
> * Run pmca-gui-v0.17-win.exe that you just downloaded
> * When GUI opens press "Get camera info" to make sure that the camera is connected. The window should display camera model and other info
> * Go to "Install app" tab
> * Make sure that "Select an app from the app list" is ticked
> * From the dropdown chose "OpenMemories: Tweak"
> * Press "Install selected app"
> * Tick "Select an apk"
> * Press "Open apk..."
> * Find "PMCA_CleanHDMI.apk" that you just downloaded
> * Press "Install selected app"
> * Disconnect the camera from the PC
> * In camera go to Menu -> Application List -> OpenMemories: Tweak -> Developer
> * Tick "Enable Telnet"
> * Press on "Wifi settings"
> * Press on "Access Point Set"
> * Choose your wifi and enter it's password
> * Go back to OpenMemories: Tweak -> Developer. Tick "Enable Wifi"
> * Wait till it connects. It will display IP address which you will need later.
> * In your PC open Settings -> search "telnet" -> click on "Turn Windows features on or off" - Select "Telnet Client" -> Press “OK"
> * Open terminal. To do that type "terminal" in Search -> click on "Windows terminal"
> * type "telnet ". For example "telnet 192.168.2.9". Press "Enter".
> * Now copy these commands in terminal:
> 
> ```
> bk.elf r 0x01070a47
> bk.elf w 0x01070a47 00
> bk.elf r 0x01070a47
> ```
> 
> * Press "Enter".
> * Take your camera. Select "Movie" mode
> * Insert HDMI cable. Now your Sony a5000 should have **clean HDMI out** with 1080p resolution!
> 
> Thanks to the community! This is great! 😄

Just tried these steps and they work 100%.

Thanks so much, been trying to figure this out for ages and this did the trick!

epiceasty, 2020-12-04:
This worked for me also. Thank you so much I really thought my sony was gonna be redundant

epiceasty, 2020-12-05:
Is there a way to save these settings so u don't have to do this each time?

sanmai, 2020-12-18:
I wish someone could figure out a way to break [the annoying "skin smoothing" feature](https://www.youtube.com/watch?v=de8EQdDo1tc) in a similar fashion.

alissonpaganini, 2020-12-18:
can I apply this to a Sony nex 3n? is there any possibility to adapt to it?

[54 more replies]