ma1co/OpenMemories-Tweak #72: Can I use this app on my video camera ax40？

frixing, 2016-06-17:


ma1co, 2016-06-17:
As far as I know, nobody has ever tried that. I think you should be able to install the app, but you probably won't be able to open it, as the AX40 doesn't have an app launcher (similar to #58).

Please try it out and let us know!

frixing, 2016-06-18:
I tried. l can't find the app.

frixing, 2016-06-18:
l want to unlock  the language setting. Any other way?

ma1co, 2016-06-25:
Have you tried the [console tool](https://github.com/ma1co/Sony-PMCA-RE/releases)? Donwload it and run `pmca install`.

frixing, 2016-06-25:
I do not known how to use it.

apychan, 2016-07-08:
Tested.
I am getting this result.
See if it might help.

https://github.com/ma1co/Sony-PMCA-RE/issues/12

frixing, 2018-02-06:
Can you add a7r3？I want to test it.