ma1co/OpenMemories-Tweak #76: Error: Network connection

likebliss, 2016-06-20:
I am getting the following issue on my a7rii with the latest firmware, using OSX 10.11.5 / latest version of safari.   Is this a common issue/fix?  Thanks! 

_Loading plugin
Plugin loaded
Starting task
Got task key
Cancelling plugin
Plugin cancelled
Starting download to camera
Error: Network connection_﻿

burado, 2016-07-04:
I am getting the same issue with my(Japanese only) a7ii with firmware V. 3.10. I have the camera set to "MTP". I am using a macbook running OS X version 10.11.5 safari version  9.1.1. Below is what I am getting. I appreciate your help. Thanks.

Loading plugin
Plugin loaded
Starting task
Got task key
Cancelling plugin
Plugin cancelled
Starting download to camera
Error: Network connection

likebliss, 2016-07-05:
Oh yeah!  I realized that I had the Japanese model as well.  I wanted this
tweak to turn off the PAL/NTSC warning.  The way I got it to work was to
simply use my old mac that hasn't been updated in awhile.  Might be worth
it to ask a friend to use their windows machine, or non-updated mac.
Worked on my 10.9 iMac no problems.

duguzuyang, 2016-09-03:
Same issue here, Windows 10 version 14393.105.

Redstorm3265, 2016-12-17:
Having the same issue with the A6000

BasicObject, 2016-12-22:
I ran into this issue on Windows 7 with my A7S but switching to IE and the correct USB mode made it work.

AdrianEddy, 2017-04-05:
What fixed it for me was connecting the camera to the USB port directly, not via USB hub

ma1co, 2017-10-30:
Closing this, a lot has changed... Please open a new issue if you still have problems.

closed 2017-10-30 by ma1co