ma1co/OpenMemories-Tweak #78: Clean HDMI Output

dakser, 2016-06-26:
It could be great to have a clean HDMI output, I have a HX400V. Great job!

jackdpeterson, 2017-10-05:
I'd be happy to work on this task. What's the process to export all known settings into a format for better research and live tweaking? If I can see some documentation on this to make things a bit more intuitive I'd be happy to start researching this at least on an NEX-6.

rdweaver, 2017-10-05:
I installed the demo app on my Sony a5000 and ran the camera view and that gave a clean HDMI out except for the header bar at the top that says camera. Modifying that would possible be a potential solution. See also: https://github.com/ma1co/OpenMemories-Tweak/issues/70

DwightKF, 2019-05-26:
Hi 

Can somebody please, please tell me exactly how to do this? I've got zero experience with this so I need to be thought step by step.

frabusas, 2020-06-26:
which settings enable a clean HDMI out with the DSC-HX400V??

paulination, 2020-07-07:
Yes. Me too I'm a complete noob. Just got a sony A5000. Can someone give me a step by step on how I could use it to stream using the stuff discussed here?

freezersharp, 2021-01-19:
I have the HX400V and I read about clean HDMI, is not possible :(
I regretted buying it.

johannesfelber, 2021-12-08:
Does somebody know the setting id for clean HDMI output on the Alpha 58 (SLT-a58)?
According to https://github.com/ma1co/OpenMemories-Tweak/issues/70 I tested setting 0x01070a47 to 00 (the factory value was 01). Unfortunately this did not change anything about the HDMI output behaviour.