ma1co/OpenMemories-Tweak #80: Send full quality to smartphone for live streaming

dakser, 2016-06-26:
Maybe it could be usefull to make live streaming being able to receive full quality video from camera on Android smartphone using the "Smart Remote Control" and then using an screencaster app to live stream the image.

Bostwickenator, 2016-06-26:
There is already a Sony API for streaming which is available to developers.
From my tests there smart remote control is already running at the highest
resolution they can support.

mungewell, 2017-10-23:
Just for clarity... there is a difference between 'LiveView' (which I believe all the cameras support) and 'Streaming' that only some cameras (officially) support. LiveView will be limited in resolution and framerate.

See table of supported functions, particularly 'Remote playback':
https://developer.sony.com/develop/cameras/

mungewell, 2017-11-17:
Further to my earlier comment, some more of the cameras also support 'LiveStreaming' direct to Ustream with a 640x360 or 1280x720 feed. 

Unfortunately it is not yet know how to send this else where. Since you can link camera to your network, you could spoof DNS and set up a server to receive the stream... but that seems hard.