ma1co/OpenMemories-Tweak #82: SuperResolution 4x for A6000

duyzoro, 2016-06-29:
I know this might be impossible to accomplish, but wouldn't it be cool if your camera can increase it's resolution by 4 while at the same time lowering digital noise?. The idea work by taking a bunch of photo continuously and merge them together to make one higher resolution image with significantly less noise ( not panorama).

Bostwickenator, 2016-06-29:
You could mean one of two things here. Temporally stacking images which
lowers noise and super sampling by moving the image sensor. Neither of
these are really tweaks so much as completely new functions. The first is
best accomplished off camera where you have lots of processing power and
RAM and the second I don't think we have control of the IS actuators for.

deveron3, 2016-06-29:
There actually *aren't* any IS actuators in the A6000, so that wouldn't work anyway.

If we get that kind of granular control of IS with the A7 series II cameras though, that would be very cool.

Bostwickenator, 2016-06-29:
Yup, you could also possibly use the lens IS but that is far less likely.

duyzoro, 2016-07-01:
I dont think IS Is needed, I mean the shaking of your hand is good enough. But yeah the processing time, I think merging those photo in camera would take a while. But wouldn't it be cool! if you could do it in camera! :D

ma1co, 2016-07-04:
You can do that in photoshop, I don't see the point of doing it directly in camera. Controlling the IS directly would be cool though, that's true...

closed 2016-07-04 by ma1co