ma1co/OpenMemories-Tweak #94: Running adb server on the camera

W1zzardTPU, 2016-08-12:
Is there any way to get adbd running on the camera?

Ideally over wifi, which would make application developing and debugging a lot easier.

I poked around in the filesystem via telnet but it seems the adbd binary is missing?

W1zzardTPU, 2016-08-13:
Added adbd binary, started it using your Shell class from my own app, but the camera crashes as soon as Android Studio tries to install an apk (replicated using "adb install").

alicepergatta, 2017-02-05:
How did you added adbd binary?
I tried, but filesystem is read-only...or i doing something wrong(

ma1co, 2017-02-18:
Here's something for you to test. Unzip the file, install the apk, open the Tweak app, Enable wifi and adb with the checkboxes. You should then be able to connect to your camera by running `adb connect <your camera's ip>`.

I only tested this on my A6000 (Android 2.3.7), I'm not sure this will run on Android 4.1.2 (the newer cameras). Please test it and let me know!

[OpenMemories-Tweak-test-adb-1.zip](https://github.com/ma1co/OpenMemories-Tweak/files/785162/OpenMemories-Tweak-test-adb-1.zip)

alicepergatta, 2017-02-18:
@ma1co 
I downloaded that file, installed apk, but when i enable adb(press checkbox) - it crashes and camera seems to be rebooting((
After that, camera did'nt worked correctly(black screen) until i removed application and formatted SD card(dont know how, but thats helped me)

Tested on Sony NEX-5R v 1.0.3

jonasjuffinger, 2017-02-20:
I installed the apk, and adb works \o/
But it always crashes after ca. 1 minute. This is the adb output:

```
01-01 00:01:01.502 588-636/? D/StateMachine: handleMessage: E msg.what=131155
01-01 00:01:01.502 588-636/? D/StateMachine: processMsg: ConnectedState
01-01 00:01:01.504 588-636/? D/WifiStateMachine: ConnectedState{ what=131155 when=-14ms arg1=1 }
01-01 00:01:01.504 588-636/? D/StateMachine: processMsg: L2ConnectedState
01-01 00:01:01.504 588-636/? D/WifiStateMachine: L2ConnectedState{ what=131155 when=-15ms arg1=1 }
01-01 00:01:01.504 588-636/? D/WifiNative-ath0: doString: SIGNAL_POLL
01-01 00:01:01.510 588-636/? D/StateMachine: handleMessage: X
01-01 00:01:04.522 588-636/? D/StateMachine: handleMessage: E msg.what=131155
01-01 00:01:04.522 588-636/? D/StateMachine: processMsg: ConnectedState
01-01 00:01:04.523 588-636/? D/WifiStateMachine: ConnectedState{ what=131155 when=-13ms arg1=1 }
01-01 00:01:04.523 588-636/? D/StateMachine: processMsg: L2ConnectedState
01-01 00:01:04.523 588-636/? D/WifiStateMachine: L2ConnectedState{ what=131155 when=-13ms arg1=1 }
01-01 00:01:04.523 588-636/? D/WifiNative-ath0: doString: SIGNAL_POLL
01-01 00:01:04.530 588-636/? D/StateMachine: handleMessage: X
01-01 00:01:07.542 588-636/? D/StateMachine: handleMessage: E msg.what=131155
01-01 00:01:07.542 588-636/? D/StateMachine: processMsg: ConnectedState
01-01 00:01:07.543 588-636/? D/WifiStateMachine: ConnectedState{ what=131155 when=-13ms arg1=1 }
01-01 00:01:07.543 588-636/? D/StateMachine: processMsg: L2ConnectedState
01-01 00:01:07.544 588-636/? D/WifiStateMachine: L2ConnectedState{ what=131155 when=-14ms arg1=1 }
01-01 00:01:07.544 588-636/? D/WifiNative-ath0: doString: SIGNAL_POLL
01-01 00:01:07.550 588-636/? D/StateMachine: handleMessage: X
01-01 00:01:10.562 588-636/? D/StateMachine: handleMessage: E msg.what=131155
01-01 00:01:10.562 588-636/? D/StateMachine: processMsg: ConnectedState
01-01 00:01:10.563 588-636/? D/WifiStateMachine: ConnectedState{ what=131155 when=-13ms arg1=1 }
01-01 00:01:10.563 588-636/? D/StateMachine: processMsg: L2ConnectedState
01-01 00:01:10.563 588-636/? D/WifiStateMachine: L2ConnectedState{ what=131155 when=-13ms arg1=1 }
01-01 00:01:10.563 588-636/? D/WifiNative-ath0: doString: SIGNAL_POLL
01-01 00:01:10.569 588-636/? D/StateMachine: handleMessage: X
01-01 00:01:13.582 588-636/? D/StateMachine: handleMessage: E msg.what=131155
01-01 00:01:13.583 588-636/? D/StateMachine: processMsg: ConnectedState
01-01 00:01:13.583 588-636/? D/WifiStateMachine: ConnectedState{ what=131155 when=-14ms arg1=1 }
01-01 00:01:13.584 588-636/? D/StateMachine: processMsg: L2ConnectedState
01-01 00:01:13.585 588-636/? D/WifiStateMachine: L2ConnectedState{ what=131155 when=-15ms arg1=1 }
01-01 00:01:13.585 588-636/? D/WifiNative-ath0: doString: SIGNAL_POLL
01-01 00:01:13.589 588-636/? D/StateMachine: handleMessage: X
01-01 00:01:13.853 588-614/? V/DAConnectionManagerService-JNI: arbiter callback: cmd=4
01-01 00:01:13.853 588-614/? D/DAConnectionManagerService-JNI: pthread cond signal - wake up a thread
01-01 00:01:13.854 588-646/? I/DAConnectionManagerService-JNI: language env isn't set
01-01 00:01:13.854 588-646/? D/DAConnectionManagerService-JNI: function : 
01-01 00:01:13.854 588-646/? D/DAConnectionManagerService-JNI: function is NULL
01-01 00:01:13.855 588-646/? D/DAConnectionManagerService: onChange: state = 4
01-01 00:01:13.856 588-646/? I/ActivityManager: START {act=android.intent.action.MAIN flg=0x10200000 cmp=com.sony.scalar.dlsys.scalaralauncher/.ScalarABlackLauncher u=0} from pid 588
01-01 00:01:13.866 197-539/? I/WsClientWrap: [osal_sync_cb][154]
01-01 00:01:13.866 197-539/? I/RIGEL: [WINDOW SYSTEM] call <WindowClient_DI*>(window)->rglDI_flushVDF()
01-01 00:01:13.983 588-598/? I/WindowManager: Aspect: 169
01-01 00:01:13.983 588-598/? I/WindowManager: Aspect: 169
01-01 00:01:13.984 588-598/? I/WindowManager: Aspect: 169
01-01 00:01:13.984 588-598/? I/WindowManager: Aspect: 169
01-01 00:01:13.989 693-693/? D/ScalarALauncher: onNewIntent intent=Intent { act=android.intent.action.MAIN flg=0x10600000 cmp=com.sony.scalar.dlsys.scalaralauncher/.ScalarABlackLauncher }
01-01 00:01:13.989 693-693/? W/ScalarALauncher: setMode mode isn't changed(BLACK)
01-01 00:01:13.990 693-693/? D/ScalarALauncher: onResume start:1487585871577
01-01 00:01:13.990 693-693/? D/ScalarALauncher: onResume Gridview requestFocus
01-01 00:01:13.990 693-693/? D/ScalarALauncher: finishAndroid Comp
01-01 00:01:13.998 588-588/? D/DAConnectionManagerService: called finishComp
01-01 00:01:13.998 588-646/? D/DAConnectionManagerService: New Finish Action has effected!
01-01 00:01:13.999 693-693/? D/ScalarALauncher: onResume end:1487585871586
01-01 00:01:14.000 588-646/? D/DAConnectionManagerService: toFinishActivities is not null
01-01 00:01:14.000 588-646/? D/DAConnectionManagerService: com.sony.scalar.app.hotspotis added.
01-01 00:01:14.000 588-646/? D/DAConnectionManagerService: com.sony.scalar.dlsys.scalaramarketis added.
01-01 00:01:14.000 588-646/? D/DAConnectionManagerService: com.sony.scalar.app.wifisettingsis added.
01-01 00:01:14.000 588-646/? D/DAConnectionManagerService: com.sony.scalar.dlsys.scalarausbdlappis added.
01-01 00:01:14.000 588-646/? D/DAConnectionManagerService: com.sony.scalar.dlsys.scalaraappmanageris added.
01-01 00:01:14.000 588-646/? D/DAConnectionManagerService: com.sony.scalar.dlsys.scalarainstalleris added.
[... cut ...]
01-01 00:01:19.622 588-636/? D/StateMachine: handleMessage: E msg.what=131155
01-01 00:01:19.622 588-636/? D/StateMachine: processMsg: ConnectedState
01-01 00:01:19.623 588-636/? D/WifiStateMachine: ConnectedState{ what=131155 when=-14ms arg1=1 }
01-01 00:01:19.623 588-636/? D/StateMachine: processMsg: L2ConnectedState
01-01 00:01:19.624 588-636/? D/WifiStateMachine: L2ConnectedState{ what=131155 when=-15ms arg1=1 }
01-01 00:01:19.624 588-636/? D/WifiNative-ath0: doString: SIGNAL_POLL
```

a6300 Android 4.1.2

ma1co, 2017-02-20:
@alicepergatta You are right, this doesn't work on gen2 devices. Currently the adbd binary for armv7 only.

@jonasjuffinger Are you sure this isn't the auto-power-off feature of your camera? Try setting it to 30 minutes in the menu.

jonasjuffinger, 2017-02-20:
I'm sorry it works now.
This is absolutely amazing, installing apps with adb and getting debug output makes the developing so much better. Thank you.

ma1co, 2017-02-21:
@alicepergatta @jonasjuffinger Here's another version. This one should work on older cameras, too. Feel free to test it!

[OpenMemories-Tweak-test-adb-2.zip](https://github.com/ma1co/OpenMemories-Tweak/files/791506/OpenMemories-Tweak-test-adb-2.zip)

alicepergatta, 2017-02-22:
@ma1co 
Now works fine!(at my NEX-5R fw: v1.0.3)
I not tested all functionality, but i successfully connected and executed shell comands)
Later i will try to debug application, and there will be more detailed report.

alicepergatta, 2017-02-22:
@ma1co 
Yesterday i connected camera to charger and go to bed.
Today turned camera on and it didn'n worked correctly - only showed viewfinder image, no any GUI.
I tried ejecting SD card and battery, but than i understood that camera starting correctly after ~1 minute.

Seems to be adbd and everything needs to be stopped correctly...or i don't know
If debuggable application crashes - camera's GUI also restarts, again after ~1 minute.

Notwithstanding, adb works correctly and now my team can comfortably debug applications)

ma1co, 2017-02-22:
@alicepergatta Yes, this is a (somewhat) known problem. But it's not adb, it's the wifi. You have to turn the wifi off before you power off your camera, otherwise it will crash while powering off. After the crash, it performs some kind of cold boot, which can take a long time, especially on old cameras.

The same happens if an android app crashes, so I would always register an `UncaughtExceptionHandler` on the main thread.

ma1co, 2017-02-22:
I've just released v0.10 which includes the adb switch.

closed 2017-02-22 by ma1co

Bostwickenator, 2017-02-24:
I've been wondering about that odd reboot behavior. Hadn't thought about it being wifi though. Thanks for the tip.