ma1co/OpenMemories-Tweak #96: Why doesn't hack work on rx10 m1?

korduo, 2016-08-14:
I tried so many times even with cmd window. But i couldn't achieve to bypass video limitation. I supposed rxm2 and m1 uses similar software. Why doesn't it work on m1? And could you find a solution for it?

redaaziouaz, 2016-09-17:
hi ,i have the same problem on my Rx10 ,to change the japenees language menu ,please any issue? ,thanks

ma1co, 2016-11-04:
The RX10 does not support android apps. So for now, you're out of luck...

closed 2016-11-04 by ma1co

ma1co, 2017-10-30:
@korduo @redaaziouaz 
There is a new experimental way to tweak settings on older cameras using the firmware update functionality. Read more about it [here](https://github.com/ma1co/Sony-PMCA-RE#tweaking-camera-settings). Keep in mind that you are probably the first person to test this on an RX10, so your mileage may vary. If you attempt anything, please let us know!

korduo, 2018-01-25:
Hi,
I wanna do it but didn't understand what to do.

31.10.2017 00:01 tarihinde ma1co yazdı:

korduo, 2018-01-25:
Rx10 is not on the list. I tried it last year but didn't work  out. You 
said firmware update functionality but i didn't understand how to do it.

25.01.2018 20:18 tarihinde Yankı Garan yazdı:

korduo, 2018-01-26:
Figured it out and it works. But be aware not to disable protection 
before unplugging from the computer. Because in that mode camera takes 
itself to macro mode and doesn't focus to anything. Thanks god it can be 
undone.

Thanks for the info finally i bypass the movie recording time.

25.01.2018 20:23 tarihinde Yankı Garan yazdı:

rik-byte, 2020-09-09:
Hi korduo or ma1co . How can i install the hack on a RX10 m1? the documentation didn't help me out. grtz