ma1co/OpenMemories-Tweak #100: Request: change temperature tolerances on temperature sensors

steve500, 2016-08-22:
Anyone with an a6300 shooting 4k knows that shooting 4k is extremely unreliable. 

I am willing to void warranties and get down to the bottom of how to get this camera to shoot 4k with less overheating issues. I have disassembled the a6300 to find that the PCB components are wrapped in plenty of copper conductive materials but I honestly believe that sony has the temperature tolerances set WAY too low or at least one sensor is tripping far too quickly.

I can be standing in a room-temperature area and be filming for 20 minutes straight without a temperature warning icon, step outside into the sun on a mildly-warm day and within 30 seconds the over-temp icon appears followed by a camera shutdown. This has rendered my 4k shooting camera USELESS over the weekend while outdoors... I'm more than willing to help find out how we can adjust these temperature tolerances and add them to this tweak app. 
-steve

steve500, 2016-08-30:
Tore into the a6300's body last night to find out what was going on in there. I found there to be a large copper heatsink over the the entire motherboard unlike the a6000's lack of thermal handling materials entirely. The heatsink was mated to the motherboard on just two chips, not the entire board as anyone would expect. I figured it could be possible that some voltage regulators that are standing alone with no thermal absorbent materials mated with them could be some of the cause to its overheating. ( Reading the previous model's a6000 service manual shows that there are thermal sensors on the voltage regulating circuits) 

I removed sony's thermal pads from the copper and replaced them with a new larger piece that I cut around components to fit. I also added two pads to the opposite side of the copper heatsink to mate it with the rear frame&body of the camera to help bring thermal energy outside of the body rather than trapping it inside as the clock ticks. 

So far, I was able to shoot with a factory sony battery to a UHS-1 card in (60mbps) 4k 24 format for 1.2 hours until the battery died. I can tell you that the body gets much warmer (not too warm to handle) and ran into no issues. This test was done indoors and will need more conclusive tests outdoors. 

While some may figure this is in the wrong location, I would love for someone to work with me and at least give us access to see live temperature data from the thermal sensors inside the camera in the sony tweak app. 

-Steve

[image]
[image]
[image]

mikeytown2, 2016-08-31:
Anyway to dump the heat into the tripod mount? That seems like a good way to move heat out of the camera case. Also see the [SmallRig Camera Cage](https://www.amazon.com/dp/B0185IAYZY/) as a way to keep your hands off.