ma1co/OpenMemories-Tweak #102: Request: Manual audio gain on A6000

LGstudio, 2016-08-26:
It would be awesome to have manual control over the A6000 over the audio gain, with some possible assignable control. I use the hot shoe zoom/gun microphone, but the audio is still on auto gain. Also some kind of audio level monitor would be nice like the a6300 has. Is this request possible for the A6000?

Also would like to get some info about the development of these kind of apps. As a SW engineer, and Android app developer, I might also help with these tweaks. How should I start making my own applicatons? Is there somewhere a guide to that?

steve500, 2016-08-30:
I second this. Becoming a developer to help in this community would be brilliant. I believe this has potential to be far more capable than Magic Lantern.

LGstudio, 2016-08-30:
Yes. I believe so. If I had time to jump into the development of this, I would be already doing it.