ma1co/OpenMemories-Tweak #105: Would be possible to record video in 4K with Sony a5100 or a6000?

marcoandrevaz, 2016-08-31:
I am whiling to contribute :)

abaeyens, 2016-09-13:
I think it's possible hardware wise. But it will take a lot of (dangerous) programming!
I've based this on the following math:
The a6000 (and a5100) does 1080p60 and (using line skipping) reads every third line.
So it reads 6016*(9/16)/3 = 1128 lines 60 times per second.
So that means it can (might be able to) read 1128*(60/24) = 2820 lines 24 times per second.
This gives us a RAW frame of 6016x2820 - or 6K with an aspect ratio of 2.133 - which is around widescreen cinema.
Writing this data as Sony .ARW files gives us a bitrate of around 6016*2820*1 = 407 megaByte/s, far too high for the sd card. But compressed to MJPG this will give only around 85megaByte/s.
And if we scale it down to 4K or lower (the BIONZ X is capable of this as we have seen in the a6300) - it should be possible to get to 35 megaByte/s recording - the maximum speed of the SD card writer.

That said, I prefer to have other, simpler things like:
1.) a smaller (adjustable) sharpening radius for video - it should be halved as it's way too large and creates ugly halos.
2.) a RAW histogram
3.) highlight metering (using the RAW data)
4.) a flatter picture profile that retains the highlights (like S-LOG2). The 'liveview grading app' does something like this.

But first of all, let us be happy with the great ma1co has already done :). We can now use Sony cameras for concert recordings, unlock languages, install the Facebook app :), upload directly to Google photos, ... Please go on with your fantastic work!

dadyal, 2016-12-19:
i would love to see my A5100 capture 4K video