ma1co/OpenMemories-Tweak #108: Request: a6000 artificial horizon

klucon2, 2016-09-13:
Hey,

first up THANK YOU SO MUCH!!!! I have a japanese model and it's been a bit cumbersome to work with it from time to time. You just made my life so much easier, thank you! :-)

If I made add a request: The NEX-6 used to have an artificial horizon and I know the a7 has one too. The a6000 unfortunately doesn't (which is the camera I currently have).  I was wondering if it could maybe be activated through your tweak? Of course if the a6000 is missing the apropirate sensor, than it won't be possible, but I dont know if that's the case!? 
Would love to see this feature again.
Keep up the good work.

Best regards,
klucon2

lukakorica, 2018-02-17:
Hi,
I am going to start the work on developing some useful apps once i get some spare time in about a week. I've also considered making an artificial horizon (seems like a simple enough problem), I think it should be possible since, i think, A6000 has an accelerometer built in because it automatically rotates portrait pictures to fit the landscape aspect ratio screen right way up.

Braxtonlzl, 2018-09-14:
Is this function still under development?