ma1co/OpenMemories-Tweak #113: Request: A7RII/A7SII 4:2:2 10/12 bit 4k 60p RAW + 1080 240p

Vap77, 2016-09-21:
I strongly believe that the software of these cameras is strong enough to support these settings, considering the magic lantern improvements on a 2012 camera. If this was possible I would pay to have it...

phacebook, 2016-10-05:
+2

puredirect, 2016-10-05:
+3

bevels, 2016-10-30:
+4

puredirect, 2017-01-18:
+5
Please say something, a few months earlier you said you working on something, are there any improvements? Thx

LeighRoy20, 2017-01-24:
This would also be awesome.

puredirect, 2018-11-06:
plus 1

puredirect, 2018-11-06:
the whole world is waiting for it since a7s2 came out :)

DukeTomlist, 2024-06-20:
+1！
I have tried using fwtool to search for file ‘backup.bin’, but the program fails during execution because the exe file does not contain ‘_winzip_’ (only '.text\x00\x00\x00', '.rdata\x00\x00', '.data\x00\x00\x00', '.rsrc\x00\x00\x00', '.reloc\x00\x00'). I am not  familiar with the organization of exe file so I couldn't make any progress.  (my camera is A7R2)
I am looking for any improvement in this project!
Thank you!